RobertCNelson/ti-linux-kernel-dev #8: can sgx_build_modules.sh  be supported?

wweiradio, 2017-01-04:
Hello,
In bb-kernel respository, there is sgx_build_modules.sh support.
I am wondering if ti linux kernel dev can also support sgx build.

Thanks,

RobertCNelson, 2017-01-04:
@wweiradio most of that script is just to deal with the ti x86/bin package.

the sgx modules for v4.4.x-ti are in the apt repo:

```
ti-sgx-ti335x-modules-`uname -r`
```

Here's how i build it..

https://github.com/rcn-ee/ti-sgx-omap5/blob/master/build-ti335x-4.4.x.sh

```
git clone https://github.com/rcn-ee/ti-sgx-omap5
cd src/
../build-ti335x-4.4.x.sh
```

Regards,

wweiradio, 2017-01-04:
thank you @RobertCNelson , i will have a try.

wweiradio, 2017-01-04:
Hi @RobertCNelson 

Maybe I did not tell you my purpose fully.

I want to have the BBB support sgx drivers and also the libraries for Qt application development.
I installed the package 'ti-sgx-ti335x-modules-4.4.30-ti-r66
Looks like not all the following are installed. Maybe not all of them were built. 

```
omaplfb.ko
bufferclass_ti.ko
pvrsrvkm.ko

```
 In addition,  i want the standard libraries, for example, libGLES_CM.so, libGLESv2.so. 
I assume the sgx_build_modules.sh  were building these things. So this is the reason I was asking how can this repository to support sgx_build_modules.sh .

Please advise how to proceed. 

```
> dpkg-query -L ti-sgx-ti335x-modules-4.4.30-ti-r66
/.
/lib
/lib/modules
/lib/modules/4.4.30-ti-r66
/lib/modules/4.4.30-ti-r66/extra
/lib/modules/4.4.30-ti-r66/extra/ti335x
/lib/modules/4.4.30-ti-r66/extra/ti335x/bc_example.ko
/lib/modules/4.4.30-ti-r66/extra/ti335x/pvrsrvkm.ko
/usr
/usr/share
/usr/share/doc
/usr/share/doc/ti-sgx-ti335x-modules-4.4.30-ti-r66
/usr/share/doc/ti-sgx-ti335x-modules-4.4.30-ti-r66/copyright
/usr/share/doc/ti-sgx-ti335x-modules-4.4.30-ti-r66/README.Debian
/usr/share/doc/ti-sgx-ti335x-modules-4.4.30-ti-r66/changelog.gz
```

wweiradio, 2017-01-04:
I managed to build, but looks like omaplfb cannot be insmod 

```
[  181.698529] omaplfb: disagrees about version of symbol module_layout
[  475.645707] omaplfb: Unknown symbol register_vsync_cb (err 0)
[  475.645829] omaplfb: disagrees about version of symbol PVRGetDisplayClassJTable
[  475.645844] omaplfb: Unknown symbol PVRGetDisplayClassJTable (err -22)
[  475.645962] omaplfb: Unknown symbol unregister_vsync_cb (err 0)
```

The reason is the following patches were not applied, because of failure. 

```
       ${git} "${DIR}/patches/sgx/0002-HACK-drm-tilcdc-add-vsync-callback-for-use-in-omaplf.patch"
        ${git} "${DIR}/patches/sgx/0003-drm-tilcdc-fix-the-ping-pong-dma-tearing-issue-seen-.patch"
        ${git} "${DIR}/patches/sgx/0004-ARM-OMAP2-Use-pdata-quirks-for-sgx-deassert_hardrese.patch"
        ${git} "${DIR}/patches/sgx/0005-ARM-dts-am33xx-add-DT-node-for-gpu.patch"](url)
```

I have not figured out the reason why the patch cannot be applied.

RobertCNelson, 2017-01-04:
@wweiradio the libraries for v4.4.x-ti are here:

http://git.ti.com/gitweb/?p=graphics/omap5-sgx-ddk-um-linux.git;a=summary

```
#one line:
git clone -b ti-img-sgx/1.14.3699939 git://git.ti.com/graphics/omap5-sgx-ddk-um-linux.git --depth=1

cd omap5-sgx-ddk-um-linux/

sudo su
export DISCIMAGE=/
export TARGET_PRODUCT=ti335x
make install
exit

#fix: libgbm.so.2 (one line:)
sudo ln -s /usr/lib/arm-linux-gnueabihf/libgbm.so.1 /usr/lib/arm-linux-gnueabihf/libgbm.so.2

sudo modprobe pvrsrvkm
sudo /usr/bin/pvrsrvctl --start --no-module
```

Those 4 patches are needed for the "old" SDK, the v4.4.x-ti branch has ti's new sdk (shared with omap5/am4/am5/dra7 products)

If you want to use the "old" SDK, stick with the v4.4.x-bone tree.

Regards,

wweiradio, 2017-01-05:
@RobertCNelson 
thank you for your information. 

> 
> Those 4 patches are needed for the "old" SDK, the v4.4.x-ti branch has ti's new sdk (shared with omap5/am4/am5/dra7 products)
> 
> If you want to use the "old" SDK, stick with the v4.4.x-bone tree.

No wonder i found some of the patches does not make sense at all.

I will have a try.

Best regards,

jadonk, 2017-09-15:
not building for me.

```
# ../build-ti335x-4.4.x.sh 
Cloning into './src'...
remote: Counting objects: 356, done.
remote: Compressing objects: 100% (246/246), done.
remote: Total 356 (delta 146), reused 238 (delta 87), pack-reused 0
Receiving objects: 100% (356/356), 975.06 KiB | 355.00 KiB/s, done.
Resolving deltas: 100% (146/146), done.
Checking connectivity... done.
patching file eurasia_km/eurasiacon/build/linux2/buildvars.mk
patching file eurasia_km/services4/srvkm/common/devicemem.c
grep: /build/buildd/linux-src/Makefile: No such file or directory
grep: /build/buildd/linux-src/Makefile: No such file or directory
grep: /build/buildd/linux-src/Makefile: No such file or directory
grep: /build/buildd/linux-src/Makefile: No such file or directory
grep: /build/buildd/linux-src/include/linux/*: No such file or directory
make[1]: Warning: File 'eurasiacon/build/linux2/bits.mk' has modification time 61 s in the future
  RM       eurasiacon/binary2_omap_linux_release/host eurasiacon/binary2_omap_linux_release/target eurasiacon/binary2_omap_linux_release/docs
make[1]: warning:  Clock skew detected.  Your build may be incomplete.
make ARCH=arm CROSS_COMPILE= KERNELDIR=/build/buildd/linux-src TARGET_PRODUCT=ti335x
grep: /build/buildd/linux-src/Makefile: No such file or directory
grep: /build/buildd/linux-src/Makefile: No such file or directory
grep: /build/buildd/linux-src/Makefile: No such file or directory
grep: /build/buildd/linux-src/Makefile: No such file or directory
grep: /build/buildd/linux-src/include/linux/*: No such file or directory
../config/core.mk:351: CROSS_COMPILE is not set. Target components will be built with the host compiler
grep: /build/buildd/linux-src/Makefile: No such file or directory
grep: /build/buildd/linux-src/Makefile: No such file or directory
grep: /build/buildd/linux-src/Makefile: No such file or directory
grep: /build/buildd/linux-src/Makefile: No such file or directory
../config/core.mk:451: $(KERNELDIR)/vmlinux does not exist. Kbuild may fail.
make: Warning: File '../common/omap4.mk' has modification time 60 s in the future
make[1]: Warning: File 'eurasiacon/build/linux2/prepare_tree.mk' has modification time 60 s in the future
make[1]: warning:  Clock skew detected.  Your build may be incomplete.
eurasiacon/build/linux2/kbuild/kbuild.mk:42: *** Directory /build/buildd/linux-src must exist.  Stop.
../config/core.mk:679: recipe for target 'build' failed
make: *** [build] Error 2
```

RobertCNelson, 2017-09-15:
@jadonk the kernel building script is really setup to be run on the builder:

Here are the builds:

http://repos.rcn-ee.net/debian/pool/main/t/ti-sgx-ti335x-modules-4.4.88-ti-r122/

You can do it on a cross machine via: (normal directory is a hard-define)

```
git clone -b ti-linux-4.4.y https://github.com/RobertCNelson/ti-linux-kernel-dev normal
git clone https://github.com/rcn-ee/ti-sgx-omap5

cd normal/
./build_kernel.sh

cd ../ti-sgx-omap5/src/
../build-ti335x-4.4.x.sh
```

(i cheated, build_kernel.sh wasn't done, so it failed to find the modules file, still builds thou)
```
voodoo@hades:/opt/github/scratch/ti-sgx-omap5/src$ ../build-ti335x-4.4.x.sh 
Cloning into './src'...
remote: Counting objects: 356, done.
remote: Compressing objects: 100% (246/246), done.
remote: Total 356 (delta 146), reused 238 (delta 87), pack-reused 0
Receiving objects: 100% (356/356), 975.06 KiB | 1.34 MiB/s, done.
Resolving deltas: 100% (146/146), done.
patching file eurasia_km/eurasiacon/build/linux2/buildvars.mk
patching file eurasia_km/services4/srvkm/common/devicemem.c
  RM       eurasiacon/binary2_omap_linux_release/host eurasiacon/binary2_omap_linux_release/target eurasiacon/binary2_omap_linux_release/docs
make ARCH=arm CROSS_COMPILE=/home/voodoo/dl/gcc/gcc-linaro-5.4.1-2017.05-x86_64_arm-linux-gnueabihf/bin/arm-linux-gnueabihf- KERNELDIR=/opt/github/scratch/ti-sgx-omap5/src/../../normal/KERNEL TARGET_PRODUCT=ti335x
../config/core.mk:451: $(KERNELDIR)/vmlinux does not exist. Kbuild may fail.

  WARNING: Symbol version dump ./Module.symvers
           is missing; modules will have no dependencies and modversions.

  LD      /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/built-in.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/env/linux/osfunc.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/env/linux/mutils.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/env/linux/mmap.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/env/linux/module.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/env/linux/pdump.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/env/linux/proc.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/env/linux/pvr_bridge_k.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/env/linux/pvr_debug.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/env/linux/mm.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/env/linux/mutex.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/env/linux/event.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/env/linux/osperproc.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/common/buffer_manager.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/common/devicemem.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/common/handle.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/common/hash.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/common/lists.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/common/mem.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/common/mem_debug.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/common/metrics.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/common/osfunc_common.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/common/pdump_common.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/common/perproc.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/common/power.o
  CC [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/services4/srvkm/common/pvrsrv.o
[... cut ...]
  CC      /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/bc_example.mod.o
  LD [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/bc_example.ko
  CC      /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/pvrsrvkm.mod.o
  LD [M]  /opt/github/scratch/ti-sgx-omap5/src/src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/pvrsrvkm.ko
```

machinekoder, 2017-11-16:
Okay, so far so good.

I have been able to install the packages from the Debian repository and also to build them myself.

However,  how can I build the libGLESv2.so and co?

machinekoder, 2017-11-17:
Now I got it. The ` git://git.ti.com/graphics/omap5-sgx-ddk-um-linux.git` repo contains the files.

I also noticed that there is a branch including 4.4, so I used following command:
`git clone -b ti-img-sgx/1.14.3699939_k4.4 git://git.ti.com/graphics/omap5-sgx-ddk-um-linux.git --depth=1`

When I try to start `pvrsrvctl`:
```
$ sudo /usr/bin/pvrsrvctl --start --no-module
/usr/bin/pvrsrvctl: error while loading shared libraries: libdrm_omap.so.1: cannot open shared object file: No such file or directory
```

The `libdrm_omap.so.1` is not included with the files from the ddk repository.

machinekoder, 2017-11-17:
`sudo apt install libdrm-omap1` did the trick.

isaacsutherland, 2017-12-01:
What do I do next after using `../build-ti355x-4.4.x.sh` to make `pvrsrvkm.ko` and `bc_example.ko`? I tried running `package-ti335x.sh` with the following results:

```
$ ../package-ti335x.sh
'.//src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/pvrsrvkm.ko' -> './pvrsrvkm.ko'
'.//src/eurasia_km/eurasiacon/binary2_omap_linux_release/target/kbuild/bc_example.ko' -> './bc_example.ko'
'.//src/eurasia_km/README' -> './README'
'.//src/eurasia_km/MIT-COPYING' -> './MIT-COPYING'
../package-ti335x.sh: line 49: equivs-build: command not found
```

Then I tried just copying the `*.ko`s via ssh, but modprobe can't seem to find them:

Build system:
```
$ tar czf sgx-kos.tar.gz bc_example.ko pvrsrvkm.ko
$ scp sgx-kos.tar.gz debian@10.92.1.107:/tmp
```

Beaglebone Black:
```
debian@arm:~$ sudo tar vxf /tmp/sgx-kos.tar.gz -C /lib/modules/4.4.91-ti-r137/extra
bc_example.ko
pvrsrvkm.ko
debian@arm:~$ sudo /sbin/modprobe pvrsrvkm
modprobe: FATAL: Module pvrsrvkm not found in directory /lib/modules/4.4.91-ti-r137
debian@arm:~$ uname -a
Linux arm 4.4.91-ti-r137 #1 SMP Thu Nov 30 21:08:12 EST 2017 armv7l GNU/Linux
```

Any advice for a noob?

RobertCNelson, 2017-12-01:
@isaacsutherland yes just run:

```
cd /opt/scripts/tools/
sudo ./update_kernel.sh
```
The sgx modules will be installed along with the kernel..

Regards,

isaacsutherland, 2017-12-01:
OK, `update_kernel.sh` gives me:
```
debian@arm:~$ tree /lib/modules/4.4.91-ti-r137/extra
/lib/modules/4.4.91-ti-r137/extra
└── ti335x
    ├── bc_example.ko
    └── pvrsrvkm.ko

1 directory, 2 files
```
My previous attempts at using `sgx_build_modules.sh` and the mainline kernel lead me to believe I also need a `omaplfb.ko` and maybe `bufferclass_ti.ko`. Where can I get those?

RobertCNelson, 2017-12-01:
those two files come from the "old" (working) fbdev sgx stack..

RobertCNelson, 2017-12-01:
Only TI has the new "v4.4.x-ti/v4.9.x-ti, bc_example.ko/pvrsrvkm.ko) driver working with this TI SDK release.

RobertCNelson, 2017-12-01:
Closing Thread:  Please contact TI directly on SGX, otherwise if you get "something" working, feel free to pass the patch..

<closing>

closed 2017-12-01 by RobertCNelson

machinekoder, 2017-12-09:
@isaacsutherland If you find a solution please let me know.