RobertCNelson/ti-linux-kernel-dev #9: CONFIG_CPU_FREQ should be disabled for Xenomai

giuliomoro, 2017-01-07:
`CONFIG_CPU_FREQ` should be disabled for Xenomai, as reported, e.g.: [here](http://xenomai.org/2014/06/configuring-for-x86-based-dual-kernels/#CPU_Frequency_scaling)
... but it is enabled in defconfig in the [xenomai-4.4 branch](
https://github.com/RobertCNelson/ti-linux-kernel-dev/blob/ti-linux-xenomai-4.4.y/patches/defconfig)

Note that when you exit `menuconfig` you get a warning reminding you that `CONFIG_CPU_FREQ` should be disabled.

RobertCNelson, 2017-01-07:
@giuliomoro i thought about that, since the v4.4.x branch shares with the beagle_x15, i didn't want to run into thermal issues on that board.  With the default governor of "performance" and as long as the user doesn't have cpufreq installed, it should be running at the full 1Ghz..

Regards,

giuliomoro, 2017-01-07:
Ok great thanks

closed 2017-01-07 by giuliomoro