RobertCNelson/ti-linux-kernel-dev #11: tools/rebuild_deb.sh rebuilds the whole kernel

ojura, 2017-02-12:
I noticed that, unlike tools/rebuild.sh, tools/rebuild_deb.sh invokes at one point `make clean` and consequently rebuilds the whole kernel, even if there were no or small changes.

 This is because in the kernel source tree, in `scripts/package/Makefile`, the `deb-pkg` target invokes `make clean`, and you are using that target in `(re)build_deb.sh`. Maybe you could use the `bindeb-pkg` target, which does not build a source tarball, and does not clean the whole tree before compiling, but rather does an incremental build.

(I asked this on the linux-kbuild mailing list [here](http://www.spinics.net/lists/linux-kbuild/msg14320.html) and they gave me that advice).

giuliomoro, 2017-05-12:
I agree with @ojura, can we use `bindeb-pkg` instead of `deb-pkg` as the default target for `rebuild_deb.sh` ?

RobertCNelson, 2017-05-12:
Sorry @giuliomoro and @ojura i completely missed reading this bug!!! Yes let me check out bindeb-pkg over the weekend. ;)

RobertCNelson, 2017-05-14:
Okay, starting the roll out:

https://github.com/RobertCNelson/stable-kernel/commit/ac76b97ea21dd2c03309b4f12b155fe18d3bea19

it'll take awhile to sync all the repo's..  It's best to use "yakbuild" for older tag's:

https://github.com/RobertCNelson/yakbuild

https://github.com/RobertCNelson/yakbuild/commit/e530344c3066a300236e4b95ca336e60f77b4867

Regards,

closed 2017-06-30 by RobertCNelson