RobertCNelson/ti-linux-kernel-dev #14: How to generate patches ??

Pillar1989, 2017-07-18:
HI Robert,
I have a curious question. which script generate those patches ? manually ?
https://github.com/RobertCNelson/ti-linux-kernel-dev/tree/ti-linux-4.9.y/patches

RobertCNelson, 2017-07-18:
Hi Baozhu,

All the patches are generated and cleaned up by git's internal tool:

```
git format-patch
```
Regards,

Pillar1989, 2017-07-19:
Yes, I know this tool .All the patches come frome ti-linux-kernel
The two repo is defferents, 
git://git.ti.com/ti-linux-kernel/ti-linux-kernel.git
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git  
How to identify one commits come from ti-linux-kernel, but not include in linux-stable ?

RobertCNelson, 2017-07-19:
Yeah, TI's repo integrates the two...  On the bright side, a good chunk of ti's commits too their v4.9.x-ti tree is either in mainline..

But sorry, direct patch by patch, is mainline, isn't mainline comparison is not available.

Regards,

Pillar1989, 2017-07-19:
so which script generate patches ?

RobertCNelson, 2017-07-19:
You can generate a raw diff via:

```
voodoo@hades:~/linux-src$ git checkout v4.9.38 -b tmp
Switched to a new branch 'tmp'
voodoo@hades:~/linux-src$ git pull --no-edit git://git.ti.com/ti-linux-kernel/ti-linux-kernel.git ti-linux-4.9.y
voodoo@hades:~/linux-src$ git diff v4.9.38...HEAD > ti-mega-patch.diff
```

or you can "try" rebasing via: (but you will have to fix "TONS" of merge's.. so it's not worth it)

```
voodoo@hades:~/linux-src$ git checkout v4.9.38 -b tmp
Switched to a new branch 'tmp'
voodoo@hades:~/linux-src$ git pull --no-edit git://git.ti.com/ti-linux-kernel/ti-linux-kernel.git ti-linux-4.9.y
voodoo@hades:~/linux-src$ git rebase v4.9.38
```
Either way, you are dealing with:
```
voodoo@hades:~/linux-src$ git describe
v4.9.38-2093-g33592189bc0c
```
2093 patches/merges, so if you want to split them out into 2093 individual patches.. have fun. ;)

Regards,

Pillar1989, 2017-07-28:
Hi, Robert, I'm busy for other things these days , let's back to this issue. Thank you  very much for help me .I want to learn about this project.
```
baozhu@server:~/studio/linux-stable$ git checkout v4.4.68 -b 68
Switched to a new branch '68'
baozhu@server:~/studio/linux-stable$ git pull  --no-edit git://git.ti.com/ti-linux-kernel/ti-linux-kernel.git ti-linux-4.4.y
remote: Counting objects: 22412, done.
remote: Compressing objects: 100% (5683/5683), done.
remote: Total 21349 (delta 18268), reused 18462 (delta 15648)
Receiving objects: 100% (21349/21349), 4.17 MiB | 251.00 KiB/s, done.
Resolving deltas: 100% (18268/18268), completed with 881 local objects.
From git://git.ti.com/ti-linux-kernel/ti-linux-kernel
 * branch            ti-linux-4.4.y -> FETCH_HEAD
Updating be90e260..1b12699
....
baozhu@server:~/studio/linux-stable$ git describe
v4.4.68-656-g72605c6
```
There are a lot of commit conflicts. But In your  project.
```
baozhu@server:~/studio/ti/ti-linux-kernel-dev/patches$ tree
...
└── x15_revc
    └── 0001-add-am57xx-beagle-x15-revc.dtb.patch

77 directories, 275 files
baozhu@server:~/studio/ti/ti-linux-kernel-dev/patches$ 
```
Only over 270 patches.  How  to split over 656 patches, then shrink and get only 270 patches ?

Pillar1989, 2017-07-31:
@RobertCNelson Please help me

RobertCNelson, 2017-07-31:
@Pillar1989 i don't split the 656 patches, i add our 270 on top of ti's tree..

v4.4.68 (stable)
+ add ti's v4.4.x-ti branch
+ add our 270 patches
= our tree

Regards,

Pillar1989, 2017-07-31:
1，Integrates linux-stable and v4.4-ti have a lot of conflicts . how to slove them?
2,   Where are the 270 patches come from?

RobertCNelson, 2017-08-01:
1:

yeah, dealing with ti's v4.4.x-ti branch can be a pain in the rear..

That's why i'm constantly changing this sha git values:

https://github.com/RobertCNelson/ti-linux-kernel-dev/blob/ti-linux-4.4.y/version.sh#L48-L50

https://github.com/RobertCNelson/ti-linux-kernel-dev/blob/ti-linux-4.4.y/patch.sh#L101-L112

2:
The patches directory is layed out, as:
```
aufs4 -> https://github.com/sfjro/aufs4-standalone -> needed for docker
backports -> backports from mainline
bbb_overlays -> kernel overlays
beaglebone -> lots of patches for differnet models
brcmfmac -> broadcom wireless
drivers -> random drivers
dtc -> backport of device tree compiler
lts44_backports -> (unused)
packaging -> *.deb fixes
pru_rpmsg -> pru fixes
quieter -> users dont like a noisy boot log.
remoteproc_by_default -> pru fixes
reverts -> reverts 
rt -> RT Linux patchset
soc -> random soc patches
x15 -> x15 support
x15_revc -> x15 revc support
```
Some of these i've been dealing with since v3.2.x, some v3.8.x, some from v3.14.x, but mostly from the v4.1.x..  some are already mainline, some will never be mainline, some are for one or two users, some have been replaced by u-boot overlays.  Easily around 10 years of work. ;)

Regards,

Pillar1989, 2017-08-01:
Ok, I know  whare the patches come from.But ti-linux and linux-stable:
```
baozhu@server:~/studio/ti/ti-linux-kernel-dev$ cat patch.sh | grep rebase
baozhu@server:~/studio/ti/ti-linux-kernel-dev$ 
```
You dit not do "rebase".  So how to integrates them together ?

closed 2017-09-06 by Pillar1989