RobertCNelson/ti-linux-kernel-dev #15: beagleboard x15 boot form ESATA

Pillar1989, 2017-09-07:
Hi 
  beagleboard x15 software seems didnt support esata yet. I have a ssd connect the board. I didnt find any sda* device

U-Boot 2017.01-00360-gc604741cb3 (Aug 11 2017 - 15:47:09 -0500), Build: jenkins-github_Bootloader-Builder-592

CPU  : DRA752-GP ES1.1
Model: TI AM5728 BeagleBoard-X15
Board: AM572x EVM REV A.20
DRAM:  2 GiB
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1

** Unable to use mmc 0:1 for loading the env **
Using default environment

setup_board_eeprom_env: am57xx_evm
SCSI:  SATA link 0 timeout.
AHCI 0001.0300 32 slots 1 ports 3 Gbps 0x1 impl SATA mode
flags: 64bit ncq stag pm led clo only pmp pio slum part ccc apst 
scanning bus for devices...
Found 0 device(s).
Net:   <ethaddr> not set. Validating first E-fuse MAC
cpsw
Press SPACE to abort autoboot in 2 seconds
usb_boot is currently disabled
scsi_boot is currently disabled
switch to partitions #0, OK
mmc0 is current device
Scanning mmc device 0
Checking for: /uEnv.txt ...
Checking for: /boot/uEnv.txt ...
466 bytes read in 15 ms (30.3 KiB/s)
Loaded environment from /boot/uEnv.txt
debug: [dtb=am57xx-beagle-x15.dtb] ...
Checking if uname_r is set in /boot/uEnv.txt ...
debug: [uname_r=4.4.83-ti-r119] ...
loading /boot/vmlinuz-4.4.83-ti-r119 ...
8664392 bytes read in 388 ms (21.3 MiB/s)
loading /boot/dtbs/4.4.83-ti-r119/am57xx-beagle-x15.dtb ...
142980 bytes read in 42 ms (3.2 MiB/s)
loading /boot/initrd.img-4.4.83-ti-r119 ...
5926369 bytes read in 275 ms (20.6 MiB/s)

root@BeagleBoard-X15:~# dmesg | grep sata
[    2.185818] ahci 4a140000.sata: SSS flag set, parallel bus scan disabled
[    2.185847] ahci 4a140000.sata: AHCI 0001.0300 32 slots 1 ports 3 Gbps 0x1 impl platform mode
[    2.185861] ahci 4a140000.sata: flags: 64bit ncq sntf stag pm led clo only pmp pio slum part ccc apst 
root@BeagleBoard-X15:~#

RobertCNelson, 2017-09-08:
@Pillar1989 sorry, been traveling this week.. Which ssd drive? Which cable?

i've been using adapters that look like this:

https://www.amazon.com/Power-Esata-eSATAp-Sata-HENGSHENG-eSATA/dp/B01FS3KP1C/ref=sr_1_9?ie=UTF8&qid=1504836757&sr=8-9&keywords=esata+to+sata

Regards,

Pillar1989, 2017-09-09:
@RobertCNelson Yes, I have bought the same Esata to SARA wire. and ssd drive : https://www.amazon.com/Faspeed-Internal-Solid-Desktop-Laptop/dp/B015ZBTTPA/ref=lp_13710602011_1_1?srs=13710602011&ie=UTF8&qid=1504948694&sr=8-1

Pillar1989, 2017-09-09:
```

** Unable to use mmc 0:1 for loading the env **
Using default environment

U-Boot 2017.01-00360-gc604741cb3 (Aug 11 2017 - 15:47:09 -0500), Build: jenkins-github_Bootloader-Builder-592

CPU  : DRA752-GP ES1.1
Model: TI AM5728 BeagleBoard-X15
Board: AM572x EVM REV A.20
DRAM:  2 GiB
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1

** Unable to use mmc 0:1 for loading the env **
Using default environment

setup_board_eeprom_env: am57xx_evm
SCSI:  SATA link 0 timeout.
AHCI 0001.0300 32 slots 1 ports 3 Gbps 0x1 impl SATA mode
flags: 64bit ncq stag pm led clo only pmp pio slum part ccc apst 
scanning bus for devices...
Found 0 device(s).
Net:   <ethaddr> not set. Validating first E-fuse MAC
cpsw
Press SPACE to abort autoboot in 2 seconds
usb_boot is currently disabled
scsi_boot is currently disabled
switch to partitions #0, OK
mmc0 is current device
Scanning mmc device 0
Checking for: /uEnv.txt ...
Checking for: /boot/uEnv.txt ...
466 bytes read in 15 ms (30.3 KiB/s)
Loaded environment from /boot/uEnv.txt
debug: [dtb=am57xx-beagle-x15.dtb] ...
Checking if uname_r is set in /boot/uEnv.txt ...
debug: [uname_r=4.4.83-ti-r119] ...
loading /boot/vmlinuz-4.4.83-ti-r119 ...
8664392 bytes read in 388 ms (21.3 MiB/s)
loading /boot/dtbs/4.4.83-ti-r119/am57xx-beagle-x15.dtb ...
142980 bytes read in 42 ms (3.2 MiB/s)
loading /boot/initrd.img-4.4.83-ti-r119 ...
5926369 bytes read in 274 ms (20.6 MiB/s)
debug: [console=ttyO2,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet] ...
debug: [bootz 0x82000000 0x88080000:5a6de1 0x88000000] ...
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
   Loading Ramdisk to 8fa59000, end 8ffffde1 ... OK
   Loading Device Tree to 8fa33000, end 8fa58e83 ... OK

Starting kernel ...
root@BeagleBoard-X15:~# dmesg 
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.83-ti-r119 (root@b2-am57xx-beagle-x15-2gb) (gcc version 6.3.0 20170516 (Debian 6.3.0-18) ) #1 SMP Thu Aug 17 06:39:24 UTC 2017
[    0.000000] CPU: ARMv7 Processor [412fc0f2] revision 2 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] Machine model: TI AM5728 BeagleBoard-X15
[    0.000000] Reserved memory: created CMA memory pool at 0x95800000, size 56 MiB
[    0.000000] Reserved memory: initialized node ipu2_cma@95800000, compatible id shared-dma-pool
[    0.000000] Reserved memory: created CMA memory pool at 0x99000000, size 64 MiB
[    0.000000] Reserved memory: initialized node dsp1_cma@99000000, compatible id shared-dma-pool
[    0.000000] Reserved memory: created CMA memory pool at 0x9d000000, size 32 MiB
[    0.000000] Reserved memory: initialized node ipu1_cma@9d000000, compatible id shared-dma-pool
[    0.000000] Reserved memory: created CMA memory pool at 0x9f000000, size 8 MiB
[    0.000000] Reserved memory: initialized node dsp2_cma@9f000000, compatible id shared-dma-pool
[    0.000000] cma: Reserved 48 MiB at 0xfc800000
[    0.000000] Memory policy: Data cache writealloc
[... 505 lines cut ...]
bus              mmcblk0p1           stdin     tty3   tty53   urandom
char             mmcblk1             stdout    tty30  tty54   v4l
console          mmcblk1boot0        tty       tty31  tty55   vcs
cpu_dma_latency  mmcblk1boot1        tty0      tty32  tty56   vcs1
cuse             mmcblk1p1           tty1      tty33  tty57   vcs2
dce-callback     mqueue              tty10     tty34  tty58   vcs3
disk             net                 tty11     tty35  tty59   vcs4
dri              network_latency     tty12     tty36  tty6    vcs5
fb0              network_throughput  tty13     tty37  tty60   vcs6
fd               null                tty14     tty38  tty61   vcsa
full             port                tty15     tty39  tty62   vcsa1
fuse             ppp                 tty16     tty4   tty63   vcsa2
hwrng            psaux               tty17     tty40  tty7    vcsa3
i2c-0            ptmx                tty18     tty41  tty8    vcsa4
i2c-2            ptp0                tty19     tty42  tty9    vcsa5
initctl          pts                 tty2      tty43  ttyGS0  vcsa6
input            random              tty20     tty44  ttyO2   vga_arbiter
kmem             rpmsg-dce           tty21     tty45  ttyS0   video0
kmsg             rtc                 tty22     tty46  ttyS1   watchdog
lightnvm         rtc0                tty23     tty47  ttyS2   watchdog0
log              rtc1                tty24     tty48  ttyS3   zero
loop-control     rtc2                tty25     tty49  ttyS4
```

[image]

RobertCNelson, 2017-09-09:
@Pillar1989 i've ordered that drive, so will check it out next week. ;)

Pillar1989, 2017-09-09:
I also try to connect the mechanical hard drive, I can hear the mechanical hard drive in operation, but in the system which can not see it

Pillar1989, 2017-09-15:
@RobertCNelson Do you receive your ssd ?  Did you ecurrence the issue ?

RobertCNelson, 2017-09-15:
@Pillar1989 well it came up:

```
root@beaglebone:~# /opt/scripts/tools/version.sh 
git:/opt/scripts/:[2ce750d881941c5189db9e189af90517e11c079f]
eeprom:[BBRDX15_C.00yywwPX15nnnn]
dogtag:[BeagleBoard.org Debian Image 2017-09-10]
kernel:[4.4.84-ti-r120]
nodejs:[v6.11.2]
pkg:[bb-cape-overlays]:[4.4.20170728.0-0rcnee1~stretch+20170728]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
root@beaglebone:~# dmesg | grep ata
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Memory: 1643860K/1898492K available (11413K kernel code, 927K rwdata, 3816K rodata, 736K init, 841K bss, 41640K reserved, 212992K cma-reserved, 1259516K highmem)
                     .data : 0xc0fa0000 - 0xc1087c30   ( 928 kB)
[    0.101656] omap_hwmod: l3_main_2 using broken dt data from ocp
[    0.533963] omap4_sram_init:Unable to allocate sram needed to handle errata I688
[    0.533972] omap4_sram_init:Unable to get sram pool needed to handle errata I688
[    1.671761] libata version 3.00 loaded.
[    2.173004] ahci 4a140000.sata: SSS flag set, parallel bus scan disabled
[    2.173033] ahci 4a140000.sata: AHCI 0001.0300 32 slots 1 ports 3 Gbps 0x1 impl platform mode
[    2.173047] ahci 4a140000.sata: flags: 64bit ncq sntf stag pm led clo only pmp pio slum part ccc apst 
[    2.174253] ata1: SATA max UDMA/133 mmio [mem 0x4a140000-0x4a1410ff] port 0x100 irq 343
[    2.987778] ata1: SATA link up 3.0 Gbps (SStatus 123 SControl 300)
[    2.987932] ata1.00: ATA-9: faspeed F510-120G, P0901A, max UDMA/133
[    2.987939] ata1.00: 234441648 sectors, multi 1: LBA48 NCQ (depth 31/32)
[    2.988510] ata1.00: configured for UDMA/133
[   33.927799] ata1.00: exception Emask 0x0 SAct 0x1 SErr 0x0 action 0x6 frozen
[   33.934899] ata1.00: failed command: READ FPDMA QUEUED
[   33.940081] ata1.00: cmd 60/08:00:38:00:00/00:00:00:00:00/40 tag 0 ncq 4096 in
[   33.954759] ata1.00: status: { DRDY }
[   33.958450] ata1: hard resetting link
[   43.967766] ata1: softreset failed (device not ready)
[   43.972850] ata1: hard resetting link
[   53.983764] ata1: softreset failed (device not ready)
[   53.988849] ata1: hard resetting link
[   64.559768] ata1: link is slow to respond, please be patient (ready=0)
[   89.031766] ata1: softreset failed (device not ready)
[   89.036852] ata1: limiting SATA link speed to 1.5 Gbps
[   89.036858] ata1: hard resetting link
[   94.223763] ata1: softreset failed (device not ready)
[   94.228848] ata1: reset failed, giving up
[   94.232883] ata1.00: disabled
[   94.232891] ata1.00: device reported invalid CHS sector 0
[   94.232911] ata1: EH complete
[   94.427083] EXT4-fs (mmcblk0p1): mounted filesystem with ordered data mode. Opts: (null)
root@beaglebone:~# ls /dev/sda
sda   sda1  sda2  
root@beaglebone:~# ls /dev/sda*
/dev/sda  /dev/sda1  /dev/sda2
```

RobertCNelson, 2017-09-15:
it really doesn't like 4.4.x:

```
root@beaglebone:~# uname -r
4.4.87-ti-r121
root@beaglebone:~# dmesg  | grep sda
[    3.002077] sd 0:0:0:0: [sda] 234441648 512-byte logical blocks: (120 GB/112 GiB)
[    3.002349] sd 0:0:0:0: [sda] Write Protect is off
[    3.002358] sd 0:0:0:0: [sda] Mode Sense: 00 3a 00 00
[    3.002429] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    3.003333]  sda: sda1 sda2
[    3.004266] sd 0:0:0:0: [sda] Attached SCSI disk
[   94.235844] sd 0:0:0:0: [sda] tag#1 FAILED Result: hostbyte=DID_BAD_TARGET driverbyte=DRIVER_OK
[   94.235857] sd 0:0:0:0: [sda] tag#1 CDB: Read(10) 28 00 00 00 00 38 00 00 08 00
[   94.235864] blk_update_request: I/O error, dev sda, sector 56
[   94.241732] sd 0:0:0:0: [sda] tag#2 FAILED Result: hostbyte=DID_BAD_TARGET driverbyte=DRIVER_OK
[   94.241741] sd 0:0:0:0: [sda] tag#2 CDB: Read(10) 28 00 00 00 00 38 00 00 08 00
[   94.241747] blk_update_request: I/O error, dev sda, sector 56
[   94.247531] Buffer I/O error on dev sda, logical block 7, async page read
[   94.258091] sd 0:0:0:0: [sda] tag#5 FAILED Result: hostbyte=DID_BAD_TARGET driverbyte=DRIVER_OK
[   94.258094] sd 0:0:0:0: [sda] tag#6 FAILED Result: hostbyte=DID_BAD_TARGET driverbyte=DRIVER_OK
[   94.258101] sd 0:0:0:0: [sda] tag#6 CDB: Read(10) 28 00 0d f9 3f 80 00 00 08 00
[   94.258105] blk_update_request: I/O error, dev sda, sector 234438528
[   94.258164] sd 0:0:0:0: [sda] tag#7 FAILED Result: hostbyte=DID_BAD_TARGET driverbyte=DRIVER_OK
[   94.258171] sd 0:0:0:0: [sda] tag#7 CDB: Read(10) 28 00 0d f9 3f 80 00 00 08 00
[   94.258174] blk_update_request: I/O error, dev sda, sector 234438528
[   94.258179] Buffer I/O error on dev sda2, logical block 29176560, async page read
[   94.278405] sd 0:0:0:0: [sda] tag#5 CDB: Read(10) 28 00 00 0f a7 80 00 00 08 00
[   94.278412] blk_update_request: I/O error, dev sda, sector 1025920
[   94.284711] sd 0:0:0:0: [sda] tag#9 FAILED Result: hostbyte=DID_BAD_TARGET driverbyte=DRIVER_OK
[   94.284720] sd 0:0:0:0: [sda] tag#9 CDB: Read(10) 28 00 00 0f a7 80 00 00 08 00
[   94.284726] blk_update_request: I/O error, dev sda, sector 1025920
[   94.290943] Buffer I/O error on dev sda1, logical block 127984, async page read
[   99.001903] sd 0:0:0:0: [sda] tag#17 FAILED Result: hostbyte=DID_BAD_TARGET driverbyte=DRIVER_OK
[   99.001918] sd 0:0:0:0: [sda] tag#17 CDB: Read(10) 28 00 0d f9 4b 00 00 00 08 00
[   99.001926] blk_update_request: I/O error, dev sda, sector 234441472
[   99.010581] sd 0:0:0:0: [sda] tag#18 FAILED Result: hostbyte=DID_BAD_TARGET driverbyte=DRIVER_OK
[   99.010600] sd 0:0:0:0: [sda] tag#18 CDB: Read(10) 28 00 0d f9 4b 00 00 00 08 00
[   99.010610] blk_update_request: I/O error, dev sda, sector 234441472
[   99.017099] Buffer I/O error on dev sda, logical block 29305184, async page read
[   99.315027] sd 0:0:0:0: [sda] tag#20 FAILED Result: hostbyte=DID_BAD_TARGET driverbyte=DRIVER_OK
[   99.315042] sd 0:0:0:0: [sda] tag#20 CDB: ATA command pass through(16) 85 06 20 00 05 00 fe 00 00 00 00 00 00 40 ef 00
[   99.319519] sd 0:0:0:0: [sda] tag#27 FAILED Result: hostbyte=DID_BAD_TARGET driverbyte=DRIVER_OK
[   99.319533] sd 0:0:0:0: [sda] tag#27 CDB: Read(10) 28 00 00 0f a7 80 00 00 08 00
[   99.319541] blk_update_request: I/O error, dev sda, sector 1025920
[   99.323530] sd 0:0:0:0: [sda] tag#28 FAILED Result: hostbyte=DID_BAD_TARGET driverbyte=DRIVER_OK
[   99.323537] sd 0:0:0:0: [sda] tag#28 CDB: Read(10) 28 00 0d f9 3f 80 00 00 08 00
[   99.323541] blk_update_request: I/O error, dev sda, sector 234438528
[   99.323867] sd 0:0:0:0: [sda] tag#29 FAILED Result: hostbyte=DID_BAD_TARGET driverbyte=DRIVER_OK
[   99.323873] sd 0:0:0:0: [sda] tag#29 CDB: Read(10) 28 00 0d f9 3f 80 00 00 08 00
[   99.323877] blk_update_request: I/O error, dev sda, sector 234438528
[   99.323881] Buffer I/O error on dev sda2, logical block 29176560, async page read
[   99.346263] sd 0:0:0:0: [sda] tag#1 FAILED Result: hostbyte=DID_BAD_TARGET driverbyte=DRIVER_OK
[   99.346275] sd 0:0:0:0: [sda] tag#1 CDB: Read(10) 28 00 00 0f a7 80 00 00 08 00
[   99.346282] blk_update_request: I/O error, dev sda, sector 1025920
[   99.352512] Buffer I/O error on dev sda1, logical block 127984, async page read
```

RobertCNelson, 2017-09-15:
it's failing in v4.9.x-ti too:

```
[  153.485094] ata1: softreset failed (device not ready)
[  153.490209] ata1: reset failed, giving up
[  153.494836] blk_update_request: I/O error, dev sda, sector 56
[  153.502978] blk_update_request: I/O error, dev sda, sector 64
[  153.508932] blk_update_request: I/O error, dev sda, sector 32
[  153.514787] blk_update_request: I/O error, dev sda, sector 16
[  153.520636] blk_update_request: I/O error, dev sda, sector 56
[  153.526436] Buffer I/O error on dev sda, logical block 7, async page read
[  153.579015] blk_update_request: I/O error, dev sda, sector 234438528
[  153.579085] blk_update_request: I/O error, dev sda, sector 1025920
[  153.579209] blk_update_request: I/O error, dev sda, sector 1025920
[  153.579215] Buffer I/O error on dev sda1, logical block 127984, async page read
[  153.606072] blk_update_request: I/O error, dev sda, sector 234438528
[  153.612510] Buffer I/O error on dev sda2, logical block 29176560, async page read
```

the drive it just junk, returning it..

Regards,

RobertCNelson, 2017-09-15:
Here is my old Patriot Pyro (120G)

```
root@beaglebone:~# uname -r
4.9.50-ti-r59
root@beaglebone:~# dmesg | grep ata
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Memory: 1640912K/1898492K available (12288K kernel code, 1109K rwdata, 4176K rodata, 1024K init, 735K bss, 44588K reserved, 212992K cma-reserved, 1259516K highmem)
                     .data : 0xc1300000 - 0xc1415770   (1110 kB)
[    0.359414] omap_hwmod: l3_main_2 using broken dt data from ocp
[    0.674250] omap4_sram_init:Unable to allocate sram needed to handle errata I688
[    0.674257] omap4_sram_init:Unable to get sram pool needed to handle errata I688
[    1.817658] libata version 3.00 loaded.
[    3.475961] ahci 4a140000.sata: SSS flag set, parallel bus scan disabled
[    3.475986] ahci 4a140000.sata: AHCI 0001.0300 32 slots 1 ports 3 Gbps 0x1 impl platform mode
[    3.475998] ahci 4a140000.sata: flags: 64bit ncq sntf stag pm led clo only pmp pio slum part ccc apst 
[    3.477452] ata1: SATA max UDMA/133 mmio [mem 0x4a140000-0x4a1410ff] port 0x100 irq 343
[    3.796647] ata1: SATA link down (SStatus 0 SControl 300)
[    4.119609] ata1: exception Emask 0x10 SAct 0x0 SErr 0x4040000 action 0xe frozen
[    4.135276] ata1: irq_stat 0x00000040, connection status changed
[    4.150177] ata1: SError: { CommWake DevExch }
[    4.154713] ata1: hard resetting link
[    5.030153] ata1: SATA link up 3.0 Gbps (SStatus 123 SControl 300)
[    5.282204] EXT4-fs (mmcblk0p1): mounted filesystem with ordered data mode. Opts: (null)
[    5.673856] ata1.00: ATA-8: Patriot Pyro, 332ABBF0, max UDMA/133
[    5.673863] ata1.00: 234441648 sectors, multi 16: LBA48 NCQ (depth 31/32)
[    5.677950] ata1.00: configured for UDMA/133
[    5.677967] ata1: EH complete
root@beaglebone:~# hdparm -tT /dev/sda

/dev/sda:
 Timing cached reads:   1804 MB in  2.00 seconds = 902.52 MB/sec
 Timing buffered disk reads: 676 MB in  3.01 seconds = 224.64 MB/sec
root@beaglebone:~# dmesg | grep ata
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Memory: 1640912K/1898492K available (12288K kernel code, 1109K rwdata, 4176K rodata, 1024K init, 735K bss, 44588K reserved, 212992K cma-reserved, 1259516K highmem)
                     .data : 0xc1300000 - 0xc1415770   (1110 kB)
[    0.359414] omap_hwmod: l3_main_2 using broken dt data from ocp
[    0.674250] omap4_sram_init:Unable to allocate sram needed to handle errata I688
[    0.674257] omap4_sram_init:Unable to get sram pool needed to handle errata I688
[    1.817658] libata version 3.00 loaded.
[    3.475961] ahci 4a140000.sata: SSS flag set, parallel bus scan disabled
[    3.475986] ahci 4a140000.sata: AHCI 0001.0300 32 slots 1 ports 3 Gbps 0x1 impl platform mode
[    3.475998] ahci 4a140000.sata: flags: 64bit ncq sntf stag pm led clo only pmp pio slum part ccc apst 
[    3.477452] ata1: SATA max UDMA/133 mmio [mem 0x4a140000-0x4a1410ff] port 0x100 irq 343
[    3.796647] ata1: SATA link down (SStatus 0 SControl 300)
[    4.119609] ata1: exception Emask 0x10 SAct 0x0 SErr 0x4040000 action 0xe frozen
[    4.135276] ata1: irq_stat 0x00000040, connection status changed
[    4.150177] ata1: SError: { CommWake DevExch }
[    4.154713] ata1: hard resetting link
[    5.030153] ata1: SATA link up 3.0 Gbps (SStatus 123 SControl 300)
[    5.282204] EXT4-fs (mmcblk0p1): mounted filesystem with ordered data mode. Opts: (null)
[    5.673856] ata1.00: ATA-8: Patriot Pyro, 332ABBF0, max UDMA/133
[    5.673863] ata1.00: 234441648 sectors, multi 16: LBA48 NCQ (depth 31/32)
[    5.677950] ata1.00: configured for UDMA/133
[    5.677967] ata1: EH complete
```

RobertCNelson, 2017-09-15:
and a ADATA SP600 (drive is way past it's read/write limit, but still works for testing)

```
root@beaglebone:~# uname -r
4.9.50-ti-r59
root@beaglebone:~# dmesg | grep ata
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Memory: 1640912K/1898492K available (12288K kernel code, 1109K rwdata, 4176K rodata, 1024K init, 735K bss, 44588K reserved, 212992K cma-reserved, 1259516K highmem)
                     .data : 0xc1300000 - 0xc1415770   (1110 kB)
[    0.362178] omap_hwmod: l3_main_2 using broken dt data from ocp
[    0.677047] omap4_sram_init:Unable to allocate sram needed to handle errata I688
[    0.677055] omap4_sram_init:Unable to get sram pool needed to handle errata I688
[    1.820605] libata version 3.00 loaded.
[    3.490960] ahci 4a140000.sata: SSS flag set, parallel bus scan disabled
[    3.490985] ahci 4a140000.sata: AHCI 0001.0300 32 slots 1 ports 3 Gbps 0x1 impl platform mode
[    3.490997] ahci 4a140000.sata: flags: 64bit ncq sntf stag pm led clo only pmp pio slum part ccc apst 
[    3.492458] ata1: SATA max UDMA/133 mmio [mem 0x4a140000-0x4a1410ff] port 0x100 irq 343
[    9.909090] ata1: SATA link up 3.0 Gbps (SStatus 123 SControl 300)
[    9.909229] ata1.00: ATA-9: ADATA SP600, 5.8, max UDMA/133
[    9.909236] ata1.00: 250069680 sectors, multi 16: LBA48 NCQ (depth 31/32)
[    9.909456] ata1.00: configured for UDMA/133
[   12.177214] EXT4-fs (mmcblk0p1): mounted filesystem with ordered data mode. Opts: (null)
root@beaglebone:~# hdparm -tT /dev/sda

/dev/sda:
 Timing cached reads:   1758 MB in  2.00 seconds = 879.68 MB/sec
 Timing buffered disk reads: 430 MB in  3.00 seconds = 143.10 MB/sec
root@beaglebone:~# dmesg | grep ata
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Memory: 1640912K/1898492K available (12288K kernel code, 1109K rwdata, 4176K rodata, 1024K init, 735K bss, 44588K reserved, 212992K cma-reserved, 1259516K highmem)
                     .data : 0xc1300000 - 0xc1415770   (1110 kB)
[    0.362178] omap_hwmod: l3_main_2 using broken dt data from ocp
[    0.677047] omap4_sram_init:Unable to allocate sram needed to handle errata I688
[    0.677055] omap4_sram_init:Unable to get sram pool needed to handle errata I688
[    1.820605] libata version 3.00 loaded.
[    3.490960] ahci 4a140000.sata: SSS flag set, parallel bus scan disabled
[    3.490985] ahci 4a140000.sata: AHCI 0001.0300 32 slots 1 ports 3 Gbps 0x1 impl platform mode
[    3.490997] ahci 4a140000.sata: flags: 64bit ncq sntf stag pm led clo only pmp pio slum part ccc apst 
[    3.492458] ata1: SATA max UDMA/133 mmio [mem 0x4a140000-0x4a1410ff] port 0x100 irq 343
[    9.909090] ata1: SATA link up 3.0 Gbps (SStatus 123 SControl 300)
[    9.909229] ata1.00: ATA-9: ADATA SP600, 5.8, max UDMA/133
[    9.909236] ata1.00: 250069680 sectors, multi 16: LBA48 NCQ (depth 31/32)
[    9.909456] ata1.00: configured for UDMA/133
[   12.177214] EXT4-fs (mmcblk0p1): mounted filesystem with ordered data mode. Opts: (null)
```

Pillar1989, 2017-09-15:
What's update between r119 and r121 ?

Pillar1989, 2017-09-16:
That's my board version:
debian@BeagleBoard-X15:/opt/scripts/device/x15$ sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[8e0b1e6f135ff3c30468f49a7a952c98be97219a]
eeprom:[AM572PM_A.2040154P550082]
dogtag:[BeagleBoard.org Debian Image 2017-08-21]
bootloader:[eMMC-(secondary)]:[/dev/mmcblk1]:[U-Boot 2016.05-00419-g80bbf03]
kernel:[4.4.88-ti-r122]
nodejs:[v6.11.2]
device-tree-override:[dtb=am57xx-beagle-x15.dtb]
pkg:[bb-cape-overlays]:[4.4.20170728.0-0rcnee1~stretch+20170728]
pkg:[bb-wl18xx-firmware]:[1.20170821-0rcnee1~stretch+20170821]
pkg:[firmware-ti-connectivity]:[20161130-3]

The board eeprom is different.  What's your board dtb file ?

Pillar1989, 2017-09-16:
I also test the ESATA cape and disks on other platform. You can see
```
[  158.772471] ata1: exception Emask 0x10 SAct 0x0 SErr 0x40d0000 action 0xe frozen
[  158.772490] ata1: irq_stat 0x00400040, connection status changed
[  158.772502] ata1: SError: { PHYRdyChg CommWake 10B8B DevExch }
[  158.772524] ata1: hard resetting link
[  162.240961] ata1: SATA link up 3.0 Gbps (SStatus 123 SControl 300)
[  162.242518] ata1.00: ATA-8: HITACHI HTS723232A7A364, EC2ZB70R, max UDMA/100
[  162.242529] ata1.00: 625142448 sectors, multi 0: LBA48 NCQ (depth 31/32), AA
[  162.244207] ata1.00: configured for UDMA/100
[  162.244229] ata1: EH complete
[  162.244757] scsi 0:0:0:0: Direct-Access     ATA      HITACHI HTS72323 B70R PQ: 0 ANSI: 5
[  162.246480] sd 0:0:0:0: Attached scsi generic sg1 type 0
[  162.246786] sd 0:0:0:0: [sdb] 625142448 512-byte logical blocks: (320 GB/298 GiB)
[  162.246967] sd 0:0:0:0: [sdb] Write Protect is off
[  162.246980] sd 0:0:0:0: [sdb] Mode Sense: 00 3a 00 00
[  162.247062] sd 0:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[  162.313109]  sdb: sdb1 sdb2 < sdb5 sdb6 sdb7 >
[  162.317794] sd 0:0:0:0: [sdb] Attached SCSI disk
[  245.985987]  sdb: sdb1
[  246.027799]  sdb: sdb1
[  281.721019] EXT4-fs (sdb1): mounted filesystem with ordered data mode. Opts: (null)
[  453.428109] ata1: exception Emask 0x10 SAct 0x0 SErr 0x90202 action 0xe frozen
[  453.428127] ata1: irq_stat 0x00400000, PHY RDY changed
[  453.428139] ata1: SError: { RecovComm Persist PHYRdyChg 10B8B }
[  453.428156] ata1: hard resetting link
[  454.150558] ata1: SATA link down (SStatus 0 SControl 300)
[  459.150717] ata1: hard resetting link
[  459.470729] ata1: SATA link down (SStatus 0 SControl 300)
[  459.470752] ata1: limiting SATA link speed to 1.5 Gbps
[  464.470886] ata1: hard resetting link
[  464.790908] ata1: SATA link down (SStatus 0 SControl 310)
[  464.790937] ata1.00: disabled
[  464.790973] ata1: EH complete
[  464.791000] ata1.00: detaching (SCSI 0:0:0:0)
[  464.802942] sd 0:0:0:0: [sdb] Synchronizing SCSI cache
[  464.803650] sd 0:0:0:0: [sdb] Synchronize Cache(10) failed: Result: hostbyte=DID_BAD_TARGET driverbyte=DRIVER_OK
[  464.803665] sd 0:0:0:0: [sdb] Stopping disk
[  464.803704] sd 0:0:0:0: [sdb] Start/Stop Unit failed: Result: hostbyte=DID_BAD_TARGET driverbyte=DRIVER_OK
[  469.049411] ata1: exception Emask 0x10 SAct 0x0 SErr 0x40c0000 action 0xe frozen
[  469.049429] ata1: irq_stat 0x00000040, connection status changed
[  469.049441] ata1: SError: { CommWake 10B8B DevExch }
[  469.049462] ata1: hard resetting link
[  469.943074] ata1: SATA link up 3.0 Gbps (SStatus 123 SControl 300)
[  469.944543] ata1.00: ATA-9: FASPEED H8-64G, P0831A, max UDMA/133
[  469.944553] ata1.00: 125045424 sectors, multi 1: LBA48 NCQ (depth 31/32), AA
[  469.951074] ata1.00: configured for UDMA/133
[  469.951096] ata1: EH complete
[  469.951655] scsi 0:0:0:0: Direct-Access     ATA      FASPEED H8-64G   1A   PQ: 0 ANSI: 5
[  469.952737] sd 0:0:0:0: [sdb] 125045424 512-byte logical blocks: (64.0 GB/59.6 GiB)
[  469.953026] sd 0:0:0:0: [sdb] Write Protect is off
[  469.953039] sd 0:0:0:0: [sdb] Mode Sense: 00 3a 00 00
[  469.953132] sd 0:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[  469.954118] sd 0:0:0:0: Attached scsi generic sg1 type 0
[  469.954408]  sdb: sdb1 sdb2
[  469.962884] sd 0:0:0:0: [sdb] Attached SCSI disk
[  493.793886]  sdb: sdb1
[  493.811910]  sdb: sdb1
[  751.156144] ata1: exception Emask 0x10 SAct 0x0 SErr 0x90202 action 0xe frozen
[  751.156163] ata1: irq_stat 0x00400000, PHY RDY changed
[  751.156175] ata1: SError: { RecovComm Persist PHYRdyChg 10B8B }
[  751.156192] ata1: hard resetting link
[  751.880349] ata1: SATA link down (SStatus 0 SControl 300)
[  756.880514] ata1: hard resetting link
[  757.200588] ata1: SATA link down (SStatus 0 SControl 300)
[  757.200618] ata1: limiting SATA link speed to 1.5 Gbps
[  762.200672] ata1: hard resetting link
[  762.520688] ata1: SATA link down (SStatus 0 SControl 310)
[  762.520711] ata1.00: disabled
[  762.520745] ata1: EH complete
[  762.520773] ata1.00: detaching (SCSI 0:0:0:0)
[  762.522704] sd 0:0:0:0: [sdb] Synchronizing SCSI cache
[  762.522825] sd 0:0:0:0: [sdb] Synchronize Cache(10) failed: Result: hostbyte=DID_BAD_TARGET driverbyte=DRIVER_OK
[  762.522837] sd 0:0:0:0: [sdb] Stopping disk
[  762.522873] sd 0:0:0:0: [sdb] Start/Stop Unit failed: Result: hostbyte=DID_BAD_TARGET driverbyte=DRIVER_OK
[  765.026869] ata1: exception Emask 0x10 SAct 0x0 SErr 0x40d0002 action 0xe frozen
[  765.026887] ata1: irq_stat 0x00000040, connection status changed
[  765.026900] ata1: SError: { RecovComm PHYRdyChg CommWake 10B8B DevExch }
[  765.026922] ata1: hard resetting link
[  767.828874] ata1: SATA link up 3.0 Gbps (SStatus 123 SControl 300)
[  767.829971] ata1.00: ATA-8: HITACHI HTS723232A7A364, EC2ZB70R, max UDMA/100
[  767.829982] ata1.00: 625142448 sectors, multi 0: LBA48 NCQ (depth 31/32), AA
[  767.831230] ata1.00: configured for UDMA/100
[  767.831252] ata1: EH complete
[  767.831786] scsi 0:0:0:0: Direct-Access     ATA      HITACHI HTS72323 B70R PQ: 0 ANSI: 5
[  767.836625] sd 0:0:0:0: [sdb] 625142448 512-byte logical blocks: (320 GB/298 GiB)
[  767.837157] sd 0:0:0:0: [sdb] Write Protect is off
[  767.837172] sd 0:0:0:0: [sdb] Mode Sense: 00 3a 00 00
[  767.837568] sd 0:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[  767.838485] sd 0:0:0:0: Attached scsi generic sg1 type 0
[  767.859744]  sdb: sdb1
[  767.868019] sd 0:0:0:0: [sdb] Attached SCSI disk

```

Pillar1989, 2017-09-22:
Thank you, I gave up . and use  hikey960 instead.

closed 2017-09-22 by Pillar1989