RobertCNelson/ti-linux-kernel-dev #16: unexpected IRQ trap at vector 91

giuliomoro, 2017-10-18:
This is on  4.4.87-ti-xenomai-r121  (773261a9edcf2ba9f609f1b66ff5d625509038d9).
Configured with: https://pastebin.com/nEk12b3y

Procedure:
```
cd /sys/class/gpio
echo 115 > export
cd gpio115
echo both > edge
```
then connect P9_27 to ground or 3.3V alternatively. You will get the following the first time it changes state:
```
[  103.992035] irq 145, desc: df19af00, depth: 0, count: 0, unhandled: 0
[  103.998529] ->handle_irq():  c0092930, handle_edge_irq+0x0/0x1a4
[  104.004587] ->irq_data.chip(): df197f10, 0xdf197f10
[  104.009498] ->action(): de0ab9c0
[  104.012738] ->action->handler(): c0623e00, gpio_sysfs_irq+0x0/0x24
[  104.018964]    IRQ_NOPROBE set
[  104.022031] unexpected IRQ trap at vector 91
[  104.026319] ------------[ cut here ]------------
[  104.030962] WARNING: CPU: 0 PID: 0 at kernel/irq/chip.c:843 __ipipe_ack_bad_irq+0x3c/0x48()
[  104.039348] Modules linked in: usb_f_midi snd_rawmidi snd_seq_device usb_f_acm u_serial usb_f_mass_storage usb_f_rndis u_ether omap_sham omap_aes_driver uio_pruss omap_rng snd_soc_davinci_mcasp rng_core snd_soc_edma snd_soc_omap snd_soc_core snd_pcm_dmaengine snd_pcm snd_timer snd soundcore spi_omap2_mcspi evdev uio_pdrv_genirq uio libcomposite ip_tables x_tables
[  104.071916] CPU: 0 PID: 0 Comm: swapper Not tainted 4.4.87-ti-xenomai-r121 #1
[  104.079082] Hardware name: Generic AM33XX (Flattened Device Tree)
[  104.085202] I-pipe domain: Linux
[  104.088470] [<c0018924>] (unwind_backtrace) from [<c0015fc4>] (show_stack+0x20/0x24)
[  104.096266] [<c0015fc4>] (show_stack) from [<c05de2ac>] (dump_stack+0x24/0x28)
[  104.103542] [<c05de2ac>] (dump_stack) from [<c003d938>] (warn_slowpath_common+0x94/0xc0)
[  104.111678] [<c003d938>] (warn_slowpath_common) from [<c003da68>] (warn_slowpath_null+0x2c/0x34)
[  104.120509] [<c003da68>] (warn_slowpath_null) from [<c0091f84>] (__ipipe_ack_bad_irq+0x3c/0x48)
[  104.129264] [<c0091f84>] (__ipipe_ack_bad_irq) from [<c00e9300>] (__ipipe_dispatch_irq+0x6c/0x210)
[  104.138274] [<c00e9300>] (__ipipe_dispatch_irq) from [<c0627a64>] (omap_gpio_irq_handler+0x19c/0x218)
[  104.147553] [<c0627a64>] (omap_gpio_irq_handler) from [<c008e564>] (handle_irq_event_percpu+0x98/0x2b8)
[  104.156996] [<c008e564>] (handle_irq_event_percpu) from [<c008e7e0>] (handle_irq_event+0x5c/0x94)
[  104.165914] [<c008e7e0>] (handle_irq_event) from [<c00928f0>] (handle_level_irq+0xd0/0x110)
[  104.174308] [<c00928f0>] (handle_level_irq) from [<c008db28>] (generic_handle_irq+0x34/0x44)
[  104.182790] [<c008db28>] (generic_handle_irq) from [<c008ddf0>] (__handle_domain_irq+0x80/0xec)
[  104.191540] [<c008ddf0>] (__handle_domain_irq) from [<c0011c14>] (handle_IRQ+0x28/0x2c)
[  104.199591] [<c0011c14>] (handle_IRQ) from [<c001cdcc>] (__ipipe_do_IRQ+0x1c/0x24)
[  104.207202] [<c001cdcc>] (__ipipe_do_IRQ) from [<c00e8bd8>] (__ipipe_do_sync_stage+0x264/0x274)
[  104.215947] [<c00e8bd8>] (__ipipe_do_sync_stage) from [<c00e8c98>] (__ipipe_do_sync_pipeline+0xb0/0xec)
[  104.225388] [<c00e8c98>] (__ipipe_do_sync_pipeline) from [<c00e9400>] (__ipipe_dispatch_irq+0x16c/0x210)
[  104.234915] [<c00e9400>] (__ipipe_dispatch_irq) from [<c00092c8>] (__ipipe_grab_irq+0x40/0x9c)
[  104.243571] [<c00092c8>] (__ipipe_grab_irq) from [<c0009738>] (omap_intc_handle_irq+0x48/0xa8)
[  104.252220] Exception stack(0xc0f5ded0 to 0xc0f5df18)
[  104.257299] dec0:                                     c00e9070 600f0013 c0f5df44 c0f5dee8
[  104.265517] dee0: c0a87514 c00096fc 00008416 00000000 00000000 c0f78d4c 00000002 c0f5c000
[  104.273733] df00: c0f5f048 c0f5f12c c0f78d4c c101e5ec c0d7c668 c0f5df44
[  104.280390] [<c0009738>] (omap_intc_handle_irq) from [<c0a87514>] (__irq_svc+0x54/0x80)
[  104.288429] Exception stack(0xc0f5dee8 to 0xc0f5df30)
[  104.293506] dee0:                   00008416 00000000 00000000 c0f78d4c 00000002 c0f5c000
[  104.301724] df00: c0f5f048 c0f5f12c c0f78d4c c101e5ec c0d7c668 c0f5df44 c00e80c4 c0f5df38
[  104.309939] df20: c00e904c c00e9070 600f0013 ffffffff
[  104.315022] [<c0a87514>] (__irq_svc) from [<c00e9070>] (ipipe_unstall_root+0x40/0x50)
[  104.322894] [<c00e9070>] (ipipe_unstall_root) from [<c0011d28>] (arch_cpu_idle+0x84/0x90)
[  104.331120] [<c0011d28>] (arch_cpu_idle) from [<c0074f04>] (default_idle_call+0x30/0x3c)
[  104.339254] [<c0074f04>] (default_idle_call) from [<c00750c0>] (cpu_startup_entry+0x1b0/0x29c)
[  104.347909] [<c00750c0>] (cpu_startup_entry) from [<c0a81b70>] (rest_init+0x94/0x98)
[  104.355700] [<c0a81b70>] (rest_init) from [<c0eebd98>] (start_kernel+0x3dc/0x3fc)
[  104.363216] ---[ end trace 403adf2bc0bf9608 ]---
[  104.367853] ------------[ cut here ]------------
[  104.372495] WARNING: CPU: 0 PID: 0 at kernel/ipipe/core.c:584 __ipipe_set_irq_pending+0xec/0x12c()
[  104.381492] Modules linked in: usb_f_midi snd_rawmidi snd_seq_device usb_f_acm u_serial usb_f_mass_storage usb_f_rndis u_ether omap_sham omap_aes_driver uio_pruss omap_rng snd_soc_davinci_mcasp rng_core snd_soc_edma snd_soc_omap snd_soc_core snd_pcm_dmaengine snd_pcm snd_timer snd soundcore spi_omap2_mcspi evdev uio_pdrv_genirq uio libcomposite ip_tables x_tables
[  104.414038] CPU: 0 PID: 0 Comm: swapper Tainted: G        W       4.4.87-ti-xenomai-r121 #1
[  104.422425] Hardware name: Generic AM33XX (Flattened Device Tree)
[  104.428543] I-pipe domain: Linux
[  104.431792] [<c0018924>] (unwind_backtrace) from [<c0015fc4>] (show_stack+0x20/0x24)
[  104.439577] [<c0015fc4>] (show_stack) from [<c05de2ac>] (dump_stack+0x24/0x28)
[  104.446841] [<c05de2ac>] (dump_stack) from [<c003d938>] (warn_slowpath_common+0x94/0xc0)
[  104.454975] [<c003d938>] (warn_slowpath_common) from [<c003da68>] (warn_slowpath_null+0x2c/0x34)
[... 19 lines cut ...]
[  104.624509] [<c0009738>] (omap_intc_handle_irq) from [<c0a87514>] (__irq_svc+0x54/0x80)
[  104.632547] Exception stack(0xc0f5dee8 to 0xc0f5df30)
[  104.637625] dee0:                   00008416 00000000 00000000 c0f78d4c 00000002 c0f5c000
[  104.645843] df00: c0f5f048 c0f5f12c c0f78d4c c101e5ec c0d7c668 c0f5df44 c00e80c4 c0f5df38
[  104.654058] df20: c00e904c c00e9070 600f0013 ffffffff
[  104.659139] [<c0a87514>] (__irq_svc) from [<c00e9070>] (ipipe_unstall_root+0x40/0x50)
[  104.667010] [<c00e9070>] (ipipe_unstall_root) from [<c0011d28>] (arch_cpu_idle+0x84/0x90)
[  104.675231] [<c0011d28>] (arch_cpu_idle) from [<c0074f04>] (default_idle_call+0x30/0x3c)
[  104.683364] [<c0074f04>] (default_idle_call) from [<c00750c0>] (cpu_startup_entry+0x1b0/0x29c)
[  104.692021] [<c00750c0>] (cpu_startup_entry) from [<c0a81b70>] (rest_init+0x94/0x98)
[  104.699805] [<c0a81b70>] (rest_init) from [<c0eebd98>] (start_kernel+0x3dc/0x3fc)
[  104.707321] ---[ end trace 403adf2bc0bf9609 ]---
```
Subsequent changes to the state of the pin give:
```
[  220.898500] irq 145, desc: df19af00, depth: 0, count: 1, unhandled: 0
[  220.904995] ->handle_irq():  c0092930, handle_edge_irq+0x0/0x1a4
[  220.911054] ->irq_data.chip(): df197f10, 0xdf197f10
[  220.915964] ->action(): de0ab9c0
[  220.919205] ->action->handler(): c0623e00, gpio_sysfs_irq+0x0/0x24
[  220.925428]    IRQ_NOPROBE set
[  220.928496] unexpected IRQ trap at vector 91
```

Not sure if this is the right place to post this issue. Please advise if I should post it to the Xenomai list.

RobertCNelson, 2017-10-18:
@giuliomoro sorry looks like a bug in xenomai:

```
debian@beaglebone:~$ uname -r
4.4.91-ti-r133
debian@beaglebone:~$ cd /sys/class/gpio/
debian@beaglebone:/sys/class/gpio$ ls
export  gpiochip0  gpiochip32  gpiochip64  gpiochip96  unexport
debian@beaglebone:/sys/class/gpio$ echo 115 > export 
debian@beaglebone:/sys/class/gpio$ cd gpio115
debian@beaglebone:/sys/class/gpio/gpio115$ ls
active_low  device  direction  edge  power  subsystem  uevent  value
debian@beaglebone:/sys/class/gpio/gpio115$ echo both > edge 

debian@beaglebone:/sys/class/gpio/gpio115$ cat value 
0
debian@beaglebone:/sys/class/gpio/gpio115$ cat value 
1
debian@beaglebone:/sys/class/gpio/gpio115$ dmesg | tail
[   23.176738] tilcdc 4830e000.lcdc: bound 0-0070 (ops tda998x_ops [tda998x])
[   23.176768] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[   23.176776] [drm] No driver support for vblank timestamp query.
[   23.190539] tilcdc 4830e000.lcdc: No connectors reported connected with modes
[   23.190579] [drm] Cannot find any crtc or sizes - going 1024x768
[   23.341770] Console: switching to colour frame buffer device 128x48
[   23.487028] tilcdc 4830e000.lcdc: fb0:  frame buffer device
[   23.487075] [drm] Initialized tilcdc 1.0.0 20121205 on minor 0
[   25.821287] [drm] Initialized pvr 1.14.3699939 20110701 on minor 1
[   27.933478] asoc-simple-card sound: i2s-hifi <-> 48038000.mcasp mapping ok
debian@beaglebone:/sys/class/gpio/gpio115$ cat value 
0
debian@beaglebone:/sys/class/gpio/gpio115$ dmesg | tail
[   23.176738] tilcdc 4830e000.lcdc: bound 0-0070 (ops tda998x_ops [tda998x])
[   23.176768] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[   23.176776] [drm] No driver support for vblank timestamp query.
[   23.190539] tilcdc 4830e000.lcdc: No connectors reported connected with modes
[   23.190579] [drm] Cannot find any crtc or sizes - going 1024x768
[   23.341770] Console: switching to colour frame buffer device 128x48
[   23.487028] tilcdc 4830e000.lcdc: fb0:  frame buffer device
[   23.487075] [drm] Initialized tilcdc 1.0.0 20121205 on minor 0
[   25.821287] [drm] Initialized pvr 1.14.3699939 20110701 on minor 1
[   27.933478] asoc-simple-card sound: i2s-hifi <-> 48038000.mcasp mapping ok
```
Ps, we migrated from xenomai to rt, hence why no more updates to the xenomai branch..

giuliomoro, 2020-05-12:
confirmed fixed as of 4.14.108-r134 . Thanks

closed 2020-05-12 by giuliomoro