RobertCNelson/ti-linux-kernel-dev #18: P9.19-P9.20.patch and PRU vs HW UART

lthiery, 2017-12-20:
In reference to [af15b78](https://github.com/RobertCNelson/ti-linux-kernel-dev/blob/ti-linux-4.9.y/patches/config_pin/0001-am335x-bone-common-config-pin-carve-out-P9.19-P9.20.patch) which you helpfully pointed out when I was asking on dtb-rebuilder, I am having a bit of trouble. 

To achieve flow control on UART1, you suggested I simply claim them:
```
config-pin P9_19 uart
config-pin P9_20 uart
```

However, I got the same old response when I attempt this
```
$ config-pin P9_19 uart
Invalid mode: uart
```

Upon closer inspection of your patch, I realized that the most relevant added pinctrl-name is in fact `pru_uart`. This is new to me but it looks like [it's software UART via the PRU peripheral](http://processors.wiki.ti.com/index.php/Soft-UART_Implementation_on_AM335X_PRU_-_Software_Users_Guide). That would be fine for my application, but there is no device file yet (/dev/ttySU0) for this kind of UART. I'm going to see how to make that happen, but I am wondering, was hardware UART intentionall omitted in this patch?

RobertCNelson, 2017-12-20:
You have something not up-to-date...

```
sudo /opt/scripts/tools/version.sh
```

Regards,

lthiery, 2017-12-20:
$ sudo /opt/scripts/tools/version.sh
```
git:/opt/scripts/:[4b0f7bb8e5966870cff0a22f1ca3a39c0239a2d5]
eeprom:[A335BNLT00C03917BBBK1DF5]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2017-12-17]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-rc1-00002-g9b09b2]
kernel:[4.9.69-ti-r84]
nodejs:[v6.12.2]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20171215.0-0rcnee1~stretch+20171215]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
dmesg | grep pinctrl-single
[    1.403415] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
END
```

RobertCNelson, 2017-12-20:
doh! i was thinking usart_tx/rx.. yeah, P9.19/P9.20 has the control signals...

```
"UART1"
uart1_rtsn
```

```
"UART1"
uart1_ctsn
```

to use them you need to use the serial binding overlay:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/devicetree/bindings/serial/serial.txt?h=v4.15-rc4

```
	uart1: serial@48022000 {
		compatible = "ti,am3352-uart", "ti,omap3-uart";
		ti,hwmods = "uart2";
		clock-frequency = <48000000>;
		reg = <0x48022000 0x2000>;
		interrupts = <73>;
		dmas = <&edma 28 0>, <&edma 29 0>;
		dma-names = "tx", "rx";
		dtr-gpios = <&gpio2 22 GPIO_ACTIVE_LOW>;
		dsr-gpios = <&gpio2 23 GPIO_ACTIVE_LOW>;
		dcd-gpios = <&gpio2 24 GPIO_ACTIVE_LOW>;
		rng-gpios = <&gpio2 25 GPIO_ACTIVE_LOW>;
		cts-gpios = <&gpio0 12 GPIO_ACTIVE_LOW>;
		rts-gpios = <&gpio0 13 GPIO_ACTIVE_LOW>;
	};

```

What you can do in your overlay that enables them ^

```
	fragment@1 {
		target = <&ocp>;
		__overlay__ {

			P9_19_pinmux {
				status = "disabled";
			};
			P9_20_pinmux {
				status = "disabled";
			};
		};
	};
```
like a remap the spi pins here:
https://github.com/RobertCNelson/bb.org-overlays/blob/master/src/arm/PB-SPI1-ETH-WIZ-CLICK.dts

Regards,

lthiery, 2017-12-20:
[I took a whack at it](https://github.com/lthiery/bb.org-overlays/commit/34c4961ade76d2485f9b1c33ef0f6a09598570b0). 

I wasn't quite able to understand the connection between your serial binding overlay example and the overlay the enables that you did. 

I think I was roughly able to understand your overlay though so I tried to work from that. As a sidenote, I have no idea what this code snippet does so I left it:
```
			channel@0 {
 				status = "disabled";
 			};
 			channel@1 {
 				status = "disabled";
 			};
```
I linked the overlay to my uEnv.txt:
`dtb_overlay=/home/debian/bb.org-overlays/src/arm/BB-UART1-FLOW-CTRL-CLICK.dtbo`

And then I rebooted. It gets found during boot and loaded but UART_TX doesn't do anything.

How can I debug? I've already checked `cat /sys/kernel/debug/pinctrl/44e10800.pinmux/pinmux | grep 'uart1'` which yields nothing.

RobertCNelson, 2017-12-20:
you forgot the cts-gpios/rts-gpios defines;

https://gist.github.com/RobertCNelson/86c351431a784adfca01014e9bcaa930

Regards,

lthiery, 2017-12-20:
Ah I appear to have messed up:
```			cts-gpios = <&gpio 12 GPIO_ACTIVE_LOW>;```
Missing the 0 after &gpio

lthiery, 2017-12-20:
So it boots properly but still not functional. A simple TX doesn't work - I have CTS pulled up to 3V.

If the overlay was doing it's job, wouldn't the pins be visible here?
```
/home/debian# sudo cat /sys/kernel/debug/pinctrl/44e10800.pinmux/pinmux-pins | grep 'uart'
pin 92 (PIN92): 44e09000.serial (GPIO UNCLAIMED) function pinmux_uart0_pins group pinmux_uart0_pins
pin 93 (PIN93): 44e09000.serial (GPIO UNCLAIMED) function pinmux_uart0_pins group pinmux_uart0_pins```

RobertCNelson, 2017-12-20:
I see them:

```
uname -r
4.14.6-ti-r17
```

```
cat /sys/kernel/debug/pinctrl/44e10800.pinmux/pinmux-pins | grep 'uart'

pin 92 (PIN92): 44e09000.serial (GPIO UNCLAIMED) function pinmux_uart0_pins group pinmux_uart0_pins
pin 93 (PIN93): 44e09000.serial (GPIO UNCLAIMED) function pinmux_uart0_pins group pinmux_uart0_pins
pin 94 (PIN94): 48022000.serial (GPIO UNCLAIMED) function pinmux_uart1_flwctrl_pins group pinmux_uart1_flwctrl_pins
pin 95 (PIN95): 48022000.serial (GPIO UNCLAIMED) function pinmux_uart1_flwctrl_pins group pinmux_uart1_flwctrl_pins
pin 96 (PIN96): 48022000.serial (GPIO UNCLAIMED) function pinmux_uart1_flwctrl_pins group pinmux_uart1_flwctrl_pins
pin 97 (PIN97): 48022000.serial (GPIO UNCLAIMED) function pinmux_uart1_flwctrl_pins group pinmux_uart1_flwctrl_pins

```

RobertCNelson, 2017-12-20:
```
/*
 * Copyright (C) 2017 Robert Nelson <robertcnelson@gmail.com>
 *
 * This program is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License version 2 as
 * published by the Free Software Foundation.
 */

/dts-v1/;
/plugin/;

#include <dt-bindings/board/am335x-bbw-bbb-base.h>
#include <dt-bindings/gpio/gpio.h>
#include <dt-bindings/pinctrl/am33xx.h>

/ {
	fragment@0 {
		target = <&am33xx_pinmux>;
		__overlay__ {
			uart1_flwctrl_pins: pinmux_uart1_flwctrl_pins {
				pinctrl-single,pins = <
					AM33XX_IOPAD(0x0984, PIN_OUTPUT_PULLDOWN | MUX_MODE0)
					AM33XX_IOPAD(0x0980, PIN_INPUT_PULLUP	 | MUX_MODE0)
					AM33XX_IOPAD(0x0978, PIN_INPUT_PULLUP	 | MUX_MODE0)
					AM33XX_IOPAD(0x097C, PIN_OUTPUT_PULLDOWN | MUX_MODE0)
				>;
			};

		};
	};

	fragment@1 {
		target = <&ocp>;
		__overlay__ {

			P9_24_pinmux {
				status = "disabled";
			};
			P9_26_pinmux {
				status = "disabled";
			};
			P9_20_pinmux {
				status = "disabled";
			};
			P9_19_pinmux {
				status = "disabled";
			};
		};
	};

	fragment@2 {
		target = <&uart1>;
		__overlay__ {

			status = "okay";
			pinctrl-names = "default";
			pinctrl-0 = <&uart1_flwctrl_pins>;
			
			#address-cells = <1>;
			#size-cells = <0>;
			
			cts-gpios = <&gpio0 12 GPIO_ACTIVE_LOW>;
			rts-gpios = <&gpio0 13 GPIO_ACTIVE_LOW>;

		};
	};
};
```

lthiery, 2017-12-20:
The issue was in fragment@2 where pinctrl-names = "uart" in mine but "default" in yours. I'm not sure where I got uart from :-S

Serial is transmitting properly but not halting when I pull CTS down. I've tested pulling it up just to make sure it's not ~CTS and the behavior never changes. I've tripled check my wiring and can upload a picture if that concerns you at all.

I did notice [in the reference that you shared](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/devicetree/bindings/serial/serial.txt?h=v4.15-rc4) that `uart-has-rtscts;` is declared. [I tried declaring it in fragment@2](https://github.com/lthiery/bb.org-overlays/commit/2204a08adf67d3599d4e69720b1d5a50b6d4b0f4) but the behavior did not change. Do you have any ideas of what I may be missing?

lthiery, 2017-12-30:
I found out that my issue was pretty silly. I was testing with screen which apparently doesn't enable HW flow control (or at least I couldn't get it to). I was able to test with Python and CTSRTS did their jobs!

closed 2017-12-30 by lthiery