RobertCNelson/ti-linux-kernel-dev #20: pinmux file not found , pin has no cape

donpadde, 2018-02-04:
Hello.
I have encountered a problem when configuring a pin. I want to use u-boot overlays.
I need to configure pins to use eQEP. Running following command:

`config-pin P8.33 qep`

results in:

`P8_33 pinmux file not found!`
`Pin has no cape: P8_33`

Here is the output of `sudo /opt/scripts/tools/version.sh`:

```
git:/opt/scripts/:[ad016da40de5805f1a6f981cbb6c454b1a7f244b]
eeprom:[A335BNLT00C02017BBBK373E]
model:[TI_AM335x_BeagleBone_Black]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-dirty]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.01-00001-g4eb802e]
kernel:[4.4.110-ti-r144]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180126.0-0rcnee0~xenial+20180126]
WARNING:pkg:[bb-wl18xx-firmware]:[NOT_INSTALLED]
WARNING:pkg:[firmware-ti-connectivity]:[NOT_INSTALLED]
groups: 'debian': no such user
groups:[]
dmesg | grep pinctrl-single
[    0.751547] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.752359] gpio-of-helper ocp:cape-universal: Allocated GPIO id=0
[    0.752513] gpio-of-helper ocp:cape-universal: Allocated GPIO id=1
[    0.752525] gpio-of-helper ocp:cape-universal: ready
END
```

Here is my uEnv.txt:

```
uname_r=4.4.110-ti-r144
enable_uboot_overlays=1
disable_uboot_overlay_video=1
enable_uboot_cape_universal=1
```

I cannot figure out what the problem is. It would be very nice if someone could help.

Patrick

RobertCNelson, 2018-02-04:
@donpadde U-Boot on the eMMC is too old:

```
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-dirty]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.01-00001-g4eb802e]
```

This should fix the issue:
```
sudo dd if=/dev/zero of=/dev/mmcblk1 count=1 seek=1 bs=128k
```
Regards,

donpadde, 2018-02-04:
Thanks @RobertCNelson for the quick reply!
Should I just ran the command and reboot the BBB?
Because after running the command 

`sudo dd if=/dev/zero of=/dev/mmcblk1 count=1 seek=1 bs=128k`

I have the same issue as mentioned above.

Patrick

RobertCNelson, 2018-02-04:
Maybe it was too old, does the bootloader still show up?

(i forget when we switch u-boot from the fat partition to being dd'ed to the drive)

Regards,

donpadde, 2018-02-04:
How can I check if the bootloader still shows up?

Patrick

RobertCNelson, 2018-02-04:
```
sudo /opt/scripts/tools/version.sh | grep boot
```

donpadde, 2018-02-04:
I think I solved my problem.
I read again your manuel on [https://eewiki.net/display/linuxonarm/BeagleBone+Black](url).
In the section "Dealing with old Bootloader in eMMC " there is a file called "uEnv.txt". I opened this file in the root-Directory on the BBB and just uncommented all lines.
After rebooting I could set P8.33 in the qep state.

@RobertCNelson is this the right way to handle my problem?

Here is the output of `sudo /opt/scripts/tools/version.sh | grep boot`:

```
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-dirty]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.01-00001-g4eb802e]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
groups: 'debian': no such user
```

RobertCNelson, 2018-02-04:
No "/uEnv.txt" isn't supported with U-Boot Overlays, it was a way for us to work with old bootloaders in the eMMC.. But just fully erase the eMMC, it's causing your issues.

donpadde, 2018-02-04:
Ok thanks for this information. It was very helpful.

Patrick

closed 2018-02-04 by donpadde