RobertCNelson/ti-linux-kernel-dev #21: Beaglebone X15: kernel 4.14.25-ti-r38 doen't boot

flebourse, 2018-03-22:
Hello,
4.14.25-ti-r38 fails to boot on the BB X15, the eMMC /dev/mmcblk1 isn't created.
4.1413-ti-r25 was fine.
Here is the log:

U-Boot SPL 2017.01-00360-gc604741cb3 (Aug 11 2017 - 15:47:09)
DRA752-GP ES2.0
Trying to boot from MMC2_2
Card did not respond to voltage select!
** Bad device mmc 0 **
Using default environment

U-Boot 2017.01-00360-gc604741cb3 (Aug 11 2017 - 15:47:09 -0500), Build: jenkins-github_Bootloader-Builder-592

CPU  : DRA752-GP ES2.0
Model: TI AM5728 BeagleBoard-X15
Board: BeagleBoard X15 REV C.00
DRAM:  2 GiB
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
MMC: no card present
** Bad device mmc 0 **
Using default environment

setup_board_eeprom_env: beagle_x15_revc
SCSI:  SATA link 0 timeout.
AHCI 0001.0300 32 slots 1 ports 3 Gbps 0x1 impl SATA mode
flags: 64bit ncq stag pm led clo only pmp pio slum part ccc apst
scanning bus for devices...
Found 0 device(s).
Net:   <ethaddr> not set. Validating first E-fuse MAC
cpsw
Press SPACE to abort autoboot in 2 seconds
usb_boot is currently disabled
scsi_boot is currently disabled
MMC: no card present
MMC: no card present
switch to partitions #0, OK
mmc1(part 0) is current device
Scanning mmc device 1
Checking for: /uEnv.txt ...
Checking for: /boot/uEnv.txt ...
541 bytes read in 5 ms (105.5 KiB/s)
Loaded environment from /boot/uEnv.txt
Checking if uname_r is set in /boot/uEnv.txt ...
debug: [uname_r=4.14.25-ti-r38] ...
loading /boot/vmlinuz-4.14.25-ti-r38 ...
11751936 bytes read in 152 ms (73.7 MiB/s)
loading /boot/dtbs/4.14.25-ti-r38/am57xx-beagle-x15-revc.dtb ...
144288 bytes read in 25 ms (5.5 MiB/s)
loading /boot/initrd.img-4.14.25-ti-r38 ...
4139418 bytes read in 60 ms (65.8 MiB/s)
debug: [console=ttyO2,115200n8 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quie\
t kgdboc=tty02,115200] ...
debug: [bootz 0x82000000 0x88080000:3f299a 0x88000000] ...
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
   Loading Ramdisk to 8fc0d000, end 8ffff99a ... OK
Starting kernel ...

[    0.002714] /cpus/cpu@0 missing clock-frequency property
[    0.002733] /cpus/cpu@1 missing clock-frequency property
[    0.669559] dmi: Firmware registration failed.
[    1.195529] omap_voltage_late_init: Voltage driver support not added
Gave up waiting for root file system device.  Common problems:
 - Boot args (cat /proc/cmdline)
   - Check rootdelay= (did the system wait long enough?)
 - Missing modules (cat /proc/modules; ls /dev)
ALERT!  /dev/mmcblk1p1 does not exist.  Dropping to a shell!
(initramfs)

-

RobertCNelson, 2018-03-22:
@flebourse okay i figured this out, in v4.14.x-ti TI switched to the new SDHC driver they backported from mainline..

Fixed via: https://github.com/RobertCNelson/ti-linux-kernel-dev/commit/8d3d9588f03707abb30981529ed250248364bae8

Regards,

flebourse, 2018-03-22:
Good, it boots now. Thank you for your quick answer.
Regards,

closed 2019-12-11 by flebourse