RobertCNelson/ti-linux-kernel-dev #25: User can't shutdown after device tree overlay bug

stashlukj, 2018-07-26:
I'm using a device tree overlay to load an fpga-region to an FPGA from my beagle X15 via Altera Passive Serial. The serial transfer works, but there is an error message, and the processor will never complete `shutdown -t now` afterwards. Am I missing a setting or step?

Someone else has had this as well:
https://archlinuxarm.org/forum/viewtopic.php?f=48&t=11416

Commands:
```
$ mkdir -p /sys/kernel/config/device-tree/overlays/cyclone
$ dd if=/lib/firmware/cyclone.dtbo of=/sys/kernel/config/device-tree/overlays/cyclone/dtbo
```

`dmesg` output:
```
[    0.000000] Linux version 4.14.56-ti-r64 (stashlukj@ee-ubuntu-jts) (gcc version 7.3.1 20180425 [linaro-7.3-2018.05 revision d29120a424ecfbc167ef90065c0eeb7f91977701] (Linaro GCC 7.3-2018.05)) #10 SMP PREEMPT Tue Jul 24 10:55:11 EDT 2018
```
...
```
[  172.767743] fpga_manager fpga0: writing RTS200-RAW.rbf to altera-ps-spi spi1.0
[  173.606113] ------------[ cut here ]------------
[  173.610762] kernel BUG at fs/sysfs/group.c:113!
[  173.615315] Internal error: Oops - BUG: 0 [#1] PREEMPT SMP ARM
[  173.621178] Modules linked in: 8021q garp mrp stp llc bnep uio_pruss fpga_region fpga_bridge pruss_soc_bus hci_uart btqca evdev iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack usb_f_acm iptable_mangle u_se
rial usb_f_ecm bluetooth iptable_filter usb_f_rndis u_ether libcomposite ecdh_generic uio_pdrv_genirq uio altera_ps_spi fpga_mgr ip_tables x_tables
[  173.655024] CPU: 1 PID: 1424 Comm: dd Not tainted 4.14.56-ti-r64 #10
[  173.661408] Hardware name: Generic DRA74X (Flattened Device Tree)
[  173.667532] task: ec001680 task.stack: ed486000
[  173.672091] PC is at internal_create_group+0x2e4/0x30c
[  173.677256] LR is at sysfs_create_groups+0x54/0x90
[  173.682071] pc : [<c0395920>]    lr : [<c0395d88>]    psr: 600b0013
[  173.688364] sp : ed487da0  ip : ed487de8  fp : ed487de4
[  173.693612] r10: ed1de0f5  r9 : 00000000  r8 : ec1735a0
[  173.698860] r7 : 00000001  r6 : ed71fa90  r5 : 00000000  r4 : 00000000
[  173.705418] r3 : 46839c87  r2 : ed71fa90  r1 : 00000000  r0 : ec1735a0
[  173.711977] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[  173.719146] Control: 10c5387d  Table: ad7b006a  DAC: 00000051
[  173.724917] Process dd (pid: 1424, stack limit = 0xed486218)
[  173.730601] Stack: (0xed487da0 to 0xed488000)
[  173.734981] 7da0: ed487dbc ed487db0 c0dc75c0 c0dc6e84 ed487dcc 00000000 c0dc7614 00000000
[  173.743202] 7dc0: ec06df00 ec1735a0 00000001 ec06df00 00000000 ed1de0f5 ed487e0c ed487de8
[  173.751421] 7de0: c0395d88 c0395648 ec173580 eeda0640 00000044 00000001 00000001 00000000
[  173.759637] 7e00: ed487ea4 ed487e10 c0ba455c c0395d40 00000000 014000c0 00000000 c0ba3124
[  173.767855] 7e20: c07c7db4 c110d43c ec173584 c15d9b70 ec1735a0 ed487e40 eeda0640 c1504dc8
[  173.776074] 7e40: 00000000 ec173598 c0ba2ee0 c0b9d3c0 00000000 c11c5d54 00000008 ed1de000
[  173.784290] 7e60: c1141570 c0ba32e8 ec06df00 00000253 00000253 46839c87 00000008 ec2f1000
[  173.792509] 7e80: 00000000 ec2f0000 00000000 ecb6f200 00000008 ecb6f200 ed487eb4 ed487ea8
[  173.800727] 7ea0: c0ba47b4 c0ba3c9c ed487ecc ed487eb8 c0b9cfd4 c0ba47a0 000000f6 ec2f1000
[  173.808946] 7ec0: ed487eec ed487ed0 c0b9d0d4 c0b9cf78 ec235ac0 ecb6f200 00000000 ed71f3c0
[  173.817163] 7ee0: ed487f14 ed487ef0 c03969a0 c0b9d078 ed71f3c0 ecb6f200 ecd51cc0 ed59fa10
[  173.825380] 7f00: 00000000 00000008 ed487f54 ed487f18 c0308d60 c0396948 00000000 00000000
[  173.833600] 7f20: ed487f4c ed71f3c8 c016eb88 00000000 ec001bc8 ec001680 c163027c ec001bec
[  173.841819] 7f40: ed71f3c0 00000000 ed487f64 ed487f58 c0308f88 c0308cd0 ed487f8c ed487f68
[  173.850036] 7f60: c01634c0 c0308f7c ffffe000 c0109224 ed487fb0 00000006 c0109224 ed486000
[  173.858254] 7f80: ed487fac ed487f90 c010d4dc c0163428 0042d258 0042d258 00000000 00000006
[  173.866470] 7fa0: 00000000 ed487fb0 c01090ac c010d414 00000000 ffffffff ffffffff 00000000
[  173.874688] 7fc0: 0042d258 0042d258 00000000 00000006 00000001 00000000 00000200 0042d1b0
[  173.882908] 7fe0: 00000000 be8984cc 00413c4b b6f32026 20080030 00000001 affae861 affaec61
[  173.891134] [<c0395920>] (internal_create_group) from [<c0395d88>] (sysfs_create_groups+0x54/0x90)
[  173.900141] [<c0395d88>] (sysfs_create_groups) from [<c0ba455c>] (__of_overlay_create+0x8cc/0xb04)
[  173.909147] [<c0ba455c>] (__of_overlay_create) from [<c0ba47b4>] (of_overlay_create+0x20/0x24)
[  173.917806] [<c0ba47b4>] (of_overlay_create) from [<c0b9cfd4>] (create_overlay+0x68/0x100)
[  173.926118] [<c0b9cfd4>] (create_overlay) from [<c0b9d0d4>] (cfs_overlay_item_dtbo_write+0x68/0xa4)
[  173.935213] [<c0b9d0d4>] (cfs_overlay_item_dtbo_write) from [<c03969a0>] (configfs_release_bin_file+0x64/0xa0)
[  173.945266] [<c03969a0>] (configfs_release_bin_file) from [<c0308d60>] (__fput+0x9c/0x234)
[  173.953575] [<c0308d60>] (__fput) from [<c0308f88>] (____fput+0x18/0x1c)
[  173.960315] [<c0308f88>] (____fput) from [<c01634c0>] (task_work_run+0xa4/0xc4)
[  173.967669] [<c01634c0>] (task_work_run) from [<c010d4dc>] (do_work_pending+0xd4/0xd8)
[  173.975632] [<c010d4dc>] (do_work_pending) from [<c01090ac>] (slow_work_pending+0xc/0x20)
[  173.983853] Code: e5941000 eaffff84 e3550000 1affff55 (e7f001f2)
[  173.989982] ---[ end trace b1fd15cd828d58ef ]---
```

RobertCNelson, 2018-07-26:
@stashlukj yeah, i need to fix overlays on the x15 by upgrading u-boot, so we can load it their and not in the kernel..  In the mean time, any luck with v4.17.x or v4.18.x?

You can grab v4.17.x via:

```bash
cd /opt/scripts/tools/
git pull
sudo ./update_kernel.sh --armv7-lpae-channel --stable
```

Regards,

stashlukj, 2018-07-26:
I'll try a more recent kernel and get back to you.

The driver documentation makes me question whether u-boot will be able to load the overlay.

from `Documentation/devicetree/bindings/fpga/fpga-region.txt`:

> An FPGA Region that exists in the live Device Tree reflects the current state.
If the live tree shows a "firmware-name" property or child nodes under a FPGA
Region, the FPGA already has been programmed.  A DTO that targets a FPGA Region
and adds the "firmware-name" property is taken as a request to reprogram the
FPGA.  After reprogramming is successful, the overlay is accepted into the live
tree.

RobertCNelson, 2018-07-26:
Yeah, correct i'm pretty sure for fpga's it has to be done in the kernel land..

PS, which kernel driver do you need, i'm pretty sure the fpga loader is currently disabled in my lpae branch but could enable that quickly..

Regards,

stashlukj, 2018-07-26:
There is no `device-tree` node in `/sys/kernel/config` for 4.17.10-armv7-lpae-x10. Is there a u-boot variable that would need to be set?

stashlukj, 2018-07-26:
I am using the `compatible = "altr,fpga-passive-serial";` driver.

RobertCNelson, 2018-07-26:
I've got a build running with these modules now enabled, should have a deb file in a few minutes..

```
CONFIG_FPGA=y
CONFIG_ALTERA_PR_IP_CORE=m
CONFIG_ALTERA_PR_IP_CORE_PLAT=m
CONFIG_FPGA_MGR_ALTERA_PS_SPI=m
CONFIG_FPGA_MGR_ALTERA_CVP=m
CONFIG_FPGA_MGR_XILINX_SPI=m
CONFIG_FPGA_MGR_ICE40_SPI=m
CONFIG_FPGA_BRIDGE=m
CONFIG_XILINX_PR_DECOUPLER=m
CONFIG_FPGA_REGION=m
CONFIG_OF_FPGA_REGION=m
```

odd i'd expect that dir to exist, rebooting a x15 now..

Regards,

RobertCNelson, 2018-07-26:
Give this a try:

```bash
wget http://rcn-ee.online/builds/linux-image-4.17.10-armv7-lpae-x10.1_1cross_armhf.deb
sudo dpkg -i linux-image-4.17.10-armv7-lpae-x10.1_1cross_armhf.deb
sudo reboot
```

Regards,

stashlukj, 2018-07-26:
That boots and runs, but there is still no `/sys/kernel/config/device-tree/overlays` node to insert the overlay.

stashlukj, 2018-08-06:
The issue still exists in v4.14.60-r67. The `CONFIG_OF_CONFIGFS` option was removed in v4.15, which explains why there is no configfs node for this in v4.17. Is there a new way to enable the configfs to do this, or are you aware of any updates to the FPGA manager to do it differently?

stashlukj, 2019-02-22:
Resolved https://github.com/RobertCNelson/ti-linux-kernel-dev/commit/a520dc4f14c617765b84eeae4c15fefc24e52a37 in pull request #28.

closed 2019-02-22 by stashlukj