RobertCNelson/ti-linux-kernel-dev #31: How to bring up-to-date a branch on this repo?

giuliomoro, 2019-03-05:
Hi there,
I am interested in reviving the `ti-linux-xenomai-4.4.y` branch (and possible other more recent Xenomai branches later on), bringing it up to date with the latest `ti-linux-4.4.y` (or whatever a suitable upstream branch is)?. Could you please explain what is the suggested workflow to do so? I looked at the scripts in `repo_maintenance`, but I could not make sense of those.
Thanks

RobertCNelson, 2019-03-05:
@giuliomoro we need to rewrite a few things, as Xenomai changed there repo's..  Our base kernel also has to be "newer" then Xenomai to make patching cleaner..

Looking at: https://gitlab.denx.de/Xenomai

looks like ipipe is good to 4.9.x

https://gitlab.denx.de/Xenomai/ipipe

and ipipe-arm is for 4.14.x+

https://gitlab.denx.de/Xenomai/ipipe-arm

That get's patched here:

https://github.com/RobertCNelson/ti-linux-kernel-dev/blob/ti-linux-xenomai-4.4.y/patch.sh#L282-L377

The main xenomai patch is from here:

https://gitlab.denx.de/Xenomai/xenomai/tree/stable/v3.0.x

So looking back at the kernels:

4.4.x:
We are at: 4.4.155: Xenomai: 4.4.176 (We are too old, so not a clean pull...)

4.9.x:
We are at: 4.9.147: Xenomai: 4.9.146 (Yay, we are not ancient, i can pull Xenomai in..)

4.14.x:
We are at: 4.14.94: Xenomai: 4.14.85 (yay, we can pull Xenomai in..)

let me look at 4.9.x & 4.14.x tomorrow..

Regards,

giuliomoro, 2019-03-05:
thanks, would you be able to detail the steps for bringing the repo up to date, as you go along, so that I can possibly try to do it myself in the future, with no need to bother you?

> 4.4.x:
> We are at: 4.4.155: Xenomai: 4.4.176 (We are too old, so not a clean pull...)

There is a 4.4.138 earlier in the commit log, if that can be of help:
https://gitlab.denx.de/Xenomai/ipipe/commit/ea7a165796165ea6e334f2b6d19a77f1853871b1

RobertCNelson, 2019-03-05:
v4.4.x is a mess..

But got v4.9.x working:

```
debian@beaglebone:~$ dmesg | grep Xenomai
[    0.915640] [Xenomai] scheduling class idle registered.
[    0.915652] [Xenomai] scheduling class rt registered.
[    0.915855] I-pipe: head domain Xenomai registered.
[    0.921395] [Xenomai] Cobalt v3.0.8 
debian@beaglebone:~$ uname -r
4.9.147-ti-xenomai-r117
```
Pushed out and now building...

RobertCNelson, 2019-03-05:
and v4.14.x-ti works too:

```
dmesgdebian@beaglebone:~$ dmesg | grep Xenomai
[    0.809238] [Xenomai] scheduling class idle registered.
[    0.809263] [Xenomai] scheduling class rt registered.
[    0.809450] I-pipe: head domain Xenomai registered.
[    0.814529] [Xenomai] Cobalt v3.0.8 
debian@beaglebone:~$ uname -r
4.14.94-ti-xenomai-r95
```

giuliomoro, 2019-03-06:
thanks, building 4.14 now. I am wondering why you [enabled RT](https://github.com/RobertCNelson/ti-linux-kernel-dev/commit/f48ec09cf0f043edb0ff295c19c31fc931c531a1)? Also, in the next commit it seems like you (at least partially?) disabled it again: https://github.com/RobertCNelson/ti-linux-kernel-dev/commit/4228879a29f7c80df5d44e9c342a74ea2e191791#diff-95d84854652267a9fbccfca9b0f08f93R354

RobertCNelson, 2019-03-06:
@giuliomoro , brain fart in the previous commit.. I had meant to fork the rt branch, but i forked the non-rt..

Then once i disabled the RT patch and enabled Xenomai, a bunch of RT-only symbols where removed like normal.. 

Regards,

giuliomoro, 2019-03-06:
It boots, but I get an error during boot:

```
root@bela:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.94-ti-xenomai-r95 (root@giulio-VirtualBox) (gcc version 7.4.1 20181213 [linaro-7.4-2019.02 revision 56ec6f6b99cc167ff0c2f8e1a2eed33b1edc85d4] (Linaro GCC 7.4-2019.02)) #1 PREEMPT Wed Mar 6 12:43:41 GMT 2019
[    0.000000] CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: TI AM335x PocketBeagle
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 131072
[    0.000000] free_area_init_node: node 0, pgdat c13d6888, node_mem_map dfb79000
[    0.000000]   Normal zone: 1152 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 131072 pages, LIFO batch:31
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] AM335X ES2.1 (sgx neon)
[    0.000000] random: get_random_bytes called from start_kernel+0xac/0x3c8 with crng_init=0
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129920
[    0.000000] Kernel command line: console=ttyS0,115200n8 root=/dev/mmcblk0p2 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 498504K/524288K available (12288K kernel code, 974K rwdata, 3928K rodata, 1024K init, 856K bss, 25784K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
                   lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0xc0008000 - 0xc0d00000   (13280 kB)
                     .init : 0xc1200000 - 0xc1300000   (1024 kB)
                     .data : 0xc1300000 - 0xc13f3964   ( 975 kB)
                      .bss : 0xc13f5000 - 0xc14cb29c   ( 857 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Tasks RCU enabled.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] IRQ: Found an INTC at 0xfa200000 (revision 5.0) with 128 interrupts
[    0.000000] OMAP clockevent source: timer2 at 24000000 Hz
[    0.000021] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 89478484971ns
[    0.000041] I-pipe, 24.000 MHz clocksource, wrap in 178956 ms
[    0.000060] clocksource: ipipe_tsc: mask: 0xffffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000082] clocksource: timer1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000090] OMAP clocksource: timer1 at 24000000 Hz
[    0.002792] timer_probe: no matching timers found
[    0.003158] Interrupt pipeline (release #6)
[    0.003499] Console: colour dummy device 80x30
[    0.003552] Calibrating delay loop... 995.32 BogoMIPS (lpj=1990656)
[    0.020742] pid_max: default: 32768 minimum: 301
[    0.020948] Security Framework initialized
[    0.020963] Yama: becoming mindful.
[    0.021007] AppArmor: AppArmor disabled by boot time parameter
[    0.021077] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.021091] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.022089] CPU: Testing write buffer coherency: ok
[    0.022160] CPU0: Spectre v2: using BPIALL workaround
[    0.023137] Setting up static identity map for 0x80100000 - 0x80100060
[    0.023342] Hierarchical SRCU implementation.
[    0.027920] devtmpfs: initialized
[    0.060797] VFP support v0.3: implementor 41 architecture 3 part 30 variant c rev 3
[    0.061284] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.061306] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.061658] xor: measuring software checksum speed
[    0.100759]    arm4regs  :  1147.000 MB/sec
[    0.140707]    8regs     :   912.000 MB/sec
[    0.180705]    32regs    :   998.000 MB/sec
[... 246 lines cut ...]
[    8.952725] usbcore: registered new interface driver rt2800usb
[   10.404962] using random self ethernet address
[   10.404982] using random host ethernet address
[   10.457396] using random self ethernet address
[   10.457417] using random host ethernet address
[   10.516450] Mass Storage Function, version: 2009/09/11
[   10.516482] LUN: removable file: (no medium)
[   11.704012] ti_am3359-tscadc 44e0d000.tscadc: chan 0 step_avg truncating to 16
[   11.704036] ti_am3359-tscadc 44e0d000.tscadc: chan 1 step_avg truncating to 16
[   11.704045] ti_am3359-tscadc 44e0d000.tscadc: chan 2 step_avg truncating to 16
[   11.704054] ti_am3359-tscadc 44e0d000.tscadc: chan 3 step_avg truncating to 16
[   11.704063] ti_am3359-tscadc 44e0d000.tscadc: chan 4 step_avg truncating to 16
[... cut ...]
[   11.704091] ti_am3359-tscadc 44e0d000.tscadc: chan 7 step_avg truncating to 16
[   12.970244] usb0: HOST MAC be:1a:ad:2a:55:01
[   12.972247] usb0: MAC be:1a:ad:2a:55:02
[   12.976169] usb1: HOST MAC be:1a:ad:2a:55:04
[   12.976547] usb1: MAC be:1a:ad:2a:55:05
[   13.211329] configfs-gadget gadget: high-speed config #1: c
[   15.755094] random: crng init done
[   15.755115] random: 7 urandom warning(s) missed due to ratelimiting
```

I guess that may be all good, except that you cannot use the PMIC driver from Xenomai?

RobertCNelson, 2019-03-06:
Better ping the Xenomai guys. ;)