RobertCNelson/ti-linux-kernel-dev #33: SGX BeagleBone/BeagleBone Black

comarius, 2019-07-25:
Hi,
I have  installed the latest kernel form ti-linux-kernel-dev
4.4.108-bone-rt-r21
The followed for sgx this link:
https://elinux.org/BeagleBoardUbuntu  section SGX Drivers
I have a usb 2 HDMI cable which leads to /dfev/fb0
I Have a 320x240 ili9342 display /dev/fb1

The procedure described there just does not work.

```
1997  sudo tar xfv GFX_5.01.01.02_es8.x.tar.gz -C /
 1998  cd /opt/gfxinstall/
 1999  sudo ./sgx-install.sh
 2000  sudo reboot
 2001  lsmod | grep omaplfb
lsmod
fb_ili9341              3507  0
snd_soc_simple_card     8529  0
uio_pruss               4309  0
tilcdc                 28087  0
snd_soc_davinci_mcasp    16406  0
snd_soc_edma            1482  1 snd_soc_davinci_mcasp
snd_soc_omap            3677  1 snd_soc_davinci_mcasp
snd_soc_core          157988  4 snd_soc_davinci_mcasp,snd_soc_edma,snd_soc_omap,snd_soc_simple_card
snd_pcm_dmaengine       5401  2 snd_soc_core,snd_soc_omap
snd_pcm                81882  4 snd_soc_davinci_mcasp,snd_soc_core,snd_soc_omap,snd_pcm_dmaengine
snd_timer              19405  1 snd_pcm
snd                    60280  3 snd_soc_core,snd_timer,snd_pcm
soundcore               7620  1 snd
evdev                  11092  1
tda998x                11320  0
fbtft_device           39334  0
8021q                  18957  0
garp                    6435  1 8021q
mrp                     7739  1 8021q
stp                     2280  1 garp
llc                     5497  2 stp,garp
usb_f_mass_storage     42605  2
usb_f_acm               7513  2
u_serial               11100  3 usb_f_acm
usb_f_ecm               9870  2
usb_f_rndis            22809  4
u_ether                12271  2 usb_f_ecm,usb_f_rndis
uio_pdrv_genirq         3981  0
uio                     9016  2 uio_pruss,uio_pdrv_genirq
libcomposite           43976  18 usb_f_acm,usb_f_ecm,usb_f_rndis,usb_f_mass_storage
iptable_nat             2125  0
nf_conntrack_ipv4      14832  1
nf_defrag_ipv4          1786  1 nf_conntrack_ipv4
nf_nat_ipv4             5648  1 iptable_nat
nf_nat                 15453  1 nf_nat_ipv4
nf_conntrack          113181  3 nf_nat,nf_nat_ipv4,nf_conntrack_ipv4
iptable_mangle          1849  0
iptable_filter          1914  0
spidev                  7865  0
tieqep                  9123  0
flexfb                 13314  0
fbtft                  32491  3 fb_ili9341,flexfb,fbtft_device
ip_tables              12312  3 iptable_filter,iptable_mangle,iptable_nat
x_tables               17737  3 ip_tables,iptable_filter,iptable_mangle
hid_cp2112             14545  8
debian@beaglebone:~$ eglinfo
Unable to initialise egl
    egl error 'EGL_BAD_ALLOC' (0x3003)
debian@beaglebone:~$ 

other info
debian@beaglebone:/sys/class/drm/card0/card0-HDMI-A-1$ cat modes 
1280x1024
1280x800
1152x864
1280x720
1280x720
1280x720
1024x768
1024x768
800x600
800x600
720x576
720x480
720x480
640x480
640x480
640x480
720x400
debian@beaglebone:/sys/class/drm/card0/card0-HDMI-A-1$ fbset

mode "1280x1024"
    geometry 1280 1024 1280 3072 16
    timings 0 0 0 0 0 0 0
    accel true
    rgba 5/11,6/5,5/0,0/0
endmode

debian@beaglebone:/sys/class/drm/card0/card0-HDMI-A-1$ eglinfo 
Unable to initialise egl
    egl error 'EGL_BAD_ALLOC' (0x3003)
debian@beaglebone:/sys/class/drm/card0/card0-HDMI-A-1$ glxgears 
Error: couldn't open display :0
debian@beaglebone:/sys/class/drm/card0/card0-HDMI-A-1$ 

```

RobertCNelson, 2019-07-25:
For SGX support, start with the "stretch-iot" image, no other image currently works out of the box..

https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Stretch_IOT_Snapshot

Regards,

comarius, 2019-07-26:
Can I apply the 108 kernel onto it ?

RobertCNelson, 2019-07-26:
Nope, you have to start with the image as-is.. it includes the sgx blob, kernel module, sgx userspace, etc..

comarius, 2019-07-26:
Nope, neither on iot image.
it does same thing. 
I am doing something or I am missing something

history

```
  9  sudo tar xfv GFX_5.01.01.02.tar.gz -C /
   10  sudo tar xfv ./GFX_5.01.01.02_es8.x.tar.gz -C /
   11  cd /opt/gfxinstall/
   12  sudo ./sgx-install.sh
   13  sudo reboot
   14  sudo apt-get update
   15  lsmod | grep omaplfb
   16  history
   17  uname -a
   18  cd /opt/scripts/tools/
   19  git pull
   20  sudo git pull
   21  sudo ./update_kernel.sh
   22  sudo reboot
   23  lsmod
   24  sudo /etc/init.d/sgx-startup.sh restart
   25  lsmod
   26  cd /opt/gfxinstall/
   27  sudo ./sgx-install.sh
   28  sudo reboot
   29  sudo apt-get install mesa-utils

```

```
debian@beaglebone:~$ uname -a
Linux beaglebone 4.14.108-ti-r109 #1 SMP PREEMPT Wed Jul 24 22:12:48 UTC 2019 armv7l GNU/Linux

ls /sys/devices/platform/ocp/56000000.sgx
driver  driver_override  drm  modalias  of_node  power  subsystem  uevent

debian@beaglebone:~$ lsmod |grep pvr
pvrsrvkm              442368  0
debian@beaglebone:~$ lsmod | grep omaplfb
debian@beaglebone:~$ 

ti-sgx-ti335x-modules-4.14.108-ti-r109 is already the newest version (1stretch).
0 upgraded, 0 newly installed, 0 to remove and 10 not upgraded.

dmesg | grep drm
[    0.954377] [drm] Initialized vgem 1.0.0 20120112 for vgem on minor 0
[    1.426793] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[    1.426799] [drm] No driver support for vblank timestamp query.
[    1.559067] [drm] Initialized tilcdc 1.0.0 20121205 for 4830e000.lcdc on minor 1
[   17.338412] [drm] Initialized pvr 1.14.3699939 20110701 for 56000000.sgx on mino

debian@beaglebone:~$ dmesg | grep sg
[    0.000000] AM335X ES2.1 (sgx neon)
[    0.901755] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 245)
[    8.519664] systemd-sysv-generator[391]: Overwriting existing symlink /run/systemd/generator.late/sgx-startup.service with real service.
[   17.338412] [drm] Initialized pvr 1.14.3699939 20110701 for 56000000.sgx on minor 2
debian@beaglebone:~$ 

```
but.

```
debian@beaglebone:~$ sgx_init_test q
----------------------- Start -----------------------
Try calling PVRSRVConnect with an invalid argument:
 OK
Call PVRSRVConnect with a valid argument:
 FAIL - PVRSRV_ERROR_INIT_FAILURE
debian@beaglebone:~$ 

```

RobertCNelson, 2019-07-26:
No! Just use the stretch-iot image "as-is" everything for sgx is already installed and enabled..

comarius, 2019-07-26:
Thx. THx for the prompt presence, always. Amazing.

closed 2019-07-26 by comarius