RobertCNelson/ti-linux-kernel-dev #35: Network issue

Davidinos, 2019-10-16:
Goodmorning to all,
I've just installed the kernel version:
Linux arm 4.14.108-ti-xenomai-r120 #1 SMP PREEMPT Thu Oct 10 09:31:35 CEST 2019 armv7l GNU/Linux
and root file system:
debian-9.8-minimal-armhf-2019-02-16
following the steps explained here: 
https://www.digikey.com/eewiki/display/linuxonarm/BeagleBone+Black

I've got some difficulties implementing an Ethercat master using SOEM repository on my beaglebone black. Reason why I've opened an issue here:
https://github.com/OpenEtherCATsociety/SOEM/issues/344
For information if I install the same repository in my PC Ubuntu 16.04, it works like a charm.

They suggest possible problems:
There has been a bug in the network stack last year that prevented outputting the correct frames. It can also be there are some firewall rules active that prevent frames returning to SOEM.

Do you have any information/ideas about that?
Thank you in advanced.

RobertCNelson, 2019-10-16:
> They suggest possible problems:
> There has been a bug in the network stack last year that prevented outputting the correct frames. It can also be there are some firewall rules active that prevent frames returning to SOEM.

Do they have a reference commit?

Regards,

Davidinos, 2019-10-17:
Hello Robert,
thank you very much for your answer. Please consider that I'm a real newbie and I'm not very practical with git :)
Below you can find the last reference: 
https://github.com/OpenEtherCATsociety/SOEM/commit/09d48acc328083f98a3652629ccf716bd5d84444

Since they think that "it looks like there are no packets returned to the beaglebone", and that ethercat deals with basic ethernet frame, could it be that there is a kernel module that need to be installed or mounted for RAW packets?
Thank you for your attention.

Davidinos, 2019-10-21:
Hello Robert,
I've made a test. I changed the kernel version down to Linux arm 4.4.113-ti-xenomai-r149.1 #1 SMP Mon Oct 21 11:55:55 CEST 2019 armv7lx, keeping debian-9.8-minimal-armhf-2019-02-16 and it works. 
So I can consider the problem as solved. If you want to deepen the differences between the two versions, please let me know.
Thank you.

Best Regards,
Davide Brunelli