RobertCNelson/ti-linux-kernel-dev #37: devices/system/cpu/cpufreq/ is empty on 4.19.73-ti-rt-r29

rvega, 2019-12-18:
I'm using this repo to compile the kernel version mentioned in the title for a beaglebone green.

I suspect that the cpu frequency is set to a low value because my app is consuming a lot more cpu than before, when I was running 4.4.x-bone-rt. 

The directory /sys/devices/system/cpu/cpufreq is empty so I don't know how to verify or set the cpu frequency.
 
Below are the relevant options from kernel defconfig.

Perhaps I am missing some kernel configs?  

dmesg does not show any message regarding cpufreq.

Please help. How can I make sure my cpu is running at full freq?

```
# CPU Frequency scaling
#
CONFIG_CPU_FREQ=y
CONFIG_CPU_FREQ_GOV_ATTR_SET=y
CONFIG_CPU_FREQ_GOV_COMMON=y
CONFIG_CPU_FREQ_STAT=y
CONFIG_CPU_FREQ_DEFAULT_GOV_PERFORMANCE=y
# CONFIG_CPU_FREQ_DEFAULT_GOV_POWERSAVE is not set
# CONFIG_CPU_FREQ_DEFAULT_GOV_USERSPACE is not set
# CONFIG_CPU_FREQ_DEFAULT_GOV_ONDEMAND is not set
# CONFIG_CPU_FREQ_DEFAULT_GOV_CONSERVATIVE is not set
# CONFIG_CPU_FREQ_DEFAULT_GOV_SCHEDUTIL is not set
CONFIG_CPU_FREQ_GOV_PERFORMANCE=y
CONFIG_CPU_FREQ_GOV_POWERSAVE=y
CONFIG_CPU_FREQ_GOV_USERSPACE=y
CONFIG_CPU_FREQ_GOV_ONDEMAND=y
CONFIG_CPU_FREQ_GOV_CONSERVATIVE=y
CONFIG_CPU_FREQ_GOV_SCHEDUTIL=y

#
# CPU frequency scaling drivers
#
CONFIG_CPUFREQ_DT=y
CONFIG_CPUFREQ_DT_PLATDEV=y
CONFIG_ARM_BIG_LITTLE_CPUFREQ=y
CONFIG_ARM_DT_BL_CPUFREQ=y
CONFIG_ARM_OMAP2PLUS_CPUFREQ=y
CONFIG_ARM_TI_CPUFREQ=y
CONFIG_QORIQ_CPUFREQ=y
```

RobertCNelson, 2019-12-18:
@rvega the kernel default support that.. figure out what you changed..

```
debian@test-bbb-5:~$ uname -r
4.19.79-ti-rt-r30
debian@test-bbb-5:~$ sudo cat /sys/devices/system/cpu/cpufreq/policy0/cpuinfo_cur_freq 
1000000
```

Regards,

closed 2019-12-18 by RobertCNelson

rvega, 2019-12-18:
Thanks for checking.