RobertCNelson/ti-linux-kernel-dev #38: linux/version.h

giuliomoro, 2020-05-12:
I noticed that `/usr/include/linux/version.h` is not part of the `linux-headers*.deb`, so that `/usr/include/linux/version.h` always is the default one that came with the distro. Is this expected behaviour?

RobertCNelson, 2020-05-12:
@giuliomoro 

Correct, debian's linux-libc-dev provides /usr/include/linux/version.h

https://packages.debian.org/buster/linux-libc-dev

There are a few tools in Debian built off this include/library.

Our headers get installed to:

```
/usr/src/`uname -r`/
```

Regards,

giuliomoro, 2020-05-12:
Thanks, so is there a standardized way of accessing the headers of the currently running kernel without using `uname` to find out their path?

RobertCNelson, 2020-05-12:
@giuliomoro this is limitation of our situation..

"sarcasm"

Feel free to send me lots of $ and i'll gladly rebuild the whole debian archive and have /usr/include/linux/version.h tied to the kernel of your choice .;)

"end/sarcasm"

We are just using Debian's userspace for our own goals, they own /usr/include/linux/version.sh  if i change it for you, i break someone else.. Best thing todo is just leave it for Debian as they have applications already built off of it pre-installed on the image.

Regards,

closed 2020-05-12 by RobertCNelson

giuliomoro, 2020-05-12:
ok thanks. As a side note, what about the libc that is built with the kernel? I tend not to install it because it breaks all the dependencies of `apt-get` ... is it supposed to be installed or not? And what issues would arise in either case?

RobertCNelson, 2020-05-12:
I also throw that file away.. Normally you could use it to replace debian's version.. But it tends to break everything.

giuliomoro, 2020-05-12:
ok thanks for confirming