RobertCNelson/ti-linux-kernel-dev #39: capemgr not working on 4.14.108-ti-r135.1

giuliomoro, 2020-05-24:
On 4.14.108-ti-r135.1  :

```
echo BB-UART4 > /sys/devices/platform/bone_capemgr/slots
```

gives

```
[   58.322660] Unable to handle kernel NULL pointer dereference at virtual address 0000000c
[   58.332744] pgd = db380000
[   58.335937] [0000000c] *pgd=9b5e8831, *pte=00000000, *ppte=00000000
[   58.344373] Internal error: Oops: 17 [#1] PREEMPT SMP ARM
[   58.349819] Modules linked in:
[   58.352901] CPU: 0 PID: 392 Comm: bash Not tainted 4.14.108-ti-r135.1 #1xross
[   58.360067] Hardware name: Generic AM33XX (Flattened Device Tree)
[   58.366188] task: dc3ede00 task.stack: db218000
[   58.370751] PC is at __of_overlay_create+0x39c/0xac8
[   58.375736] LR is at __of_overlay_create+0x3a8/0xac8
[   58.380721] pc : [<c0a88944>]    lr : [<c0a88950>]    psr: 200f0013
[   58.387014] sp : db219d58  ip : db24ae00  fp : db219dec
[   58.392260] r10: dc5a7851  r9 : 00000000  r8 : 00000004
[   58.397507] r7 : db661c80  r6 : 00000004  r5 : db337ccc  r4 : 00000110
[   58.404062] r3 : c0a87c00  r2 : 00000000  r1 : 0000002f  r0 : 00000000
[   58.410619] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[   58.417785] Control: 10c5387d  Table: 9b380019  DAC: 00000051
[   58.423555] Process bash (pid: 392, stack limit = 0xdb218218)
[   58.429325] Stack: (0xdb219d58 to 0xdb21a000)
[   58.433702] 9d40:                                                       00000000 014000c0
[   58.441920] 9d60: 00000000 c0a87a3c db219d94 db219d78 db661ca0 db661c84 c13c4818 200f0013
[   58.450136] 9d80: db219dac c1305e08 c0a7c158 db661c98 db219dac db219da0 00000000 c1087990
[   58.458354] 9da0: 00000014 dc5a7000 c1011734 c0a87c00 db390010 dc5bba18 000000a1 1cfd5bf3
[   58.466570] 9dc0: 00000002 db390010 dc5bba18 00000000 c1305e08 00000000 dc5bba18 db39025b
[   58.474787] 9de0: db219dfc db219df0 c0a89090 c0a885b4 db219e74 db219e00 c08cb3ac c0a8907c
[   58.483003] 9e00: 00000002 db219e10 c0389368 c08cc410 00000000 db39025b c1305e08 dc5bba20
[   58.491220] 9e20: 00000003 dc242010 00000003 dc5bba10 dc5a7000 dc242010 dc242010 e03b00c8
[   58.499437] 9e40: c0899fa4 1cfd5bf3 db219e70 db225a00 c1305e08 dc5bba10 db225840 dc242010
[   58.507654] 9e60: 00000000 db390010 db219eb4 db219e78 c08cc738 c08cb158 00000000 db292240
[   58.515871] 9e80: c1012fb0 1cfd5bf3 00000000 c08cc5e8 dc5c6700 00000000 00000000 db219f68
[   58.524087] 9ea0: db225840 dc5c6710 db219ecc db219eb8 c0898390 c08cc5f4 c0898368 dc5c6700
[   58.532304] 9ec0: db219ee4 db219ed0 c0387f5c c0898374 00000009 dc5c6700 db219f1c db219ee8
[   58.540521] 9ee0: c0387648 c0387f20 00000000 00000000 dc68d988 c0387550 db292240 020d0408
[   58.548738] 9f00: db219f68 00000000 020d0408 00000009 db219f34 db219f20 c02fcfc8 c038755c
[   58.556955] 9f20: 00000009 db292240 db219f64 db219f38 c02fd1d0 c02fcfac db292240 c031fd18
[   58.565171] 9f40: c1305e08 db292240 00000000 00000000 db292240 020d0408 db219fa4 db219f68
[   58.573388] 9f60: c02fd44c c02fd128 00000000 00000000 c031fbd8 1cfd5bf3 00000000 00000009
[   58.581604] 9f80: 020d0408 b6f81d60 00000004 c01090a4 db218000 00000004 00000000 db219fa8
[   58.589821] 9fa0: c0108ea0 c02fd3fc 00000009 020d0408 00000001 020d0408 00000009 00000000
[   58.598038] 9fc0: 00000009 020d0408 b6f81d60 00000004 020d0408 00000009 00000000 00000000
[   58.606254] 9fe0: 00000000 bee7291c b6ee41bb b6f20cf6 00000030 00000001 00000000 00000000
[   58.614487] [<c0a88944>] (__of_overlay_create) from [<c0a89090>] (of_overlay_create+0x20/0x24)
[   58.623155] [<c0a89090>] (of_overlay_create) from [<c08cb3ac>] (capemgr_load_slot+0x260/0x5d4)
[   58.631812] [<c08cb3ac>] (capemgr_load_slot) from [<c08cc738>] (slots_store+0x150/0x328)
[   58.639949] [<c08cc738>] (slots_store) from [<c0898390>] (dev_attr_store+0x28/0x34)
[   58.647653] [<c0898390>] (dev_attr_store) from [<c0387f5c>] (sysfs_kf_write+0x48/0x54)
[   58.655611] [<c0387f5c>] (sysfs_kf_write) from [<c0387648>] (kernfs_fop_write+0xf8/0x1dc)
[   58.663837] [<c0387648>] (kernfs_fop_write) from [<c02fcfc8>] (__vfs_write+0x28/0x48)
[   58.671709] [<c02fcfc8>] (__vfs_write) from [<c02fd1d0>] (vfs_write+0xb4/0x1c4)
[   58.679054] [<c02fd1d0>] (vfs_write) from [<c02fd44c>] (SyS_write+0x5c/0xbc)
[   58.686147] [<c02fd44c>] (SyS_write) from [<c0108ea0>] (ret_fast_syscall+0x0/0x54)
[   58.693757] Code: e5859038 e585503c e5853034 e5859028 (e590a00c)
[   58.708080] ---[ end trace a1d58a886c119ce5 ]---
```

Is this expected? Has the `capemgr` reached EOL?

RobertCNelson, 2020-05-24:
Hi @giuliomoro sorry we are no longer supporting kernel overlays, as it became a support nightmare with no mainline support/assistance. Please convert to u-boot overlays.

https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays

Regards,

giuliomoro, 2020-05-24:
That's great, we have already been using uboot overlays for a while.

Then I am wondering why the capemgr is still enabled in the config you are shipping? Is there anything that depends on it that makes it worth keeping it enabled?

RobertCNelson, 2020-05-24:
It's gone as of 4.15.x, it's only in v4.14.x for compatibility where it still worked.

giuliomoro, 2020-05-24:
ok thanks

closed 2020-05-24 by giuliomoro