RobertCNelson/ti-linux-kernel-dev #40: multible definitions yylloc

blueForestIcarus, 2020-06-04:
Branch: ti-linux-4.14.y
release: r135

failure while linking scripts/lex/__
seems to have been fixed upstream?
I belive the problem is exclusive to gcc10/clang10 or higher.

relevant logs:

>   HOSTCC  scripts/dtc/dtc-lexer.lex.o
>   HOSTCC  scripts/dtc/dtc-parser.tab.o
>   HOSTLD  scripts/dtc/fdtoverlay
>   HOSTLD  scripts/dtc/dtc
> /usr/bin/ld: scripts/dtc/dtc-parser.tab.o:(.bss+0x10): multiple definition of `yylloc'; scripts/dtc/dtc-lexer.lex.o:(.bss+0x0): first defined here
> collect2: error: ld returned 1 exit status
> make[3]: *** [scripts/Makefile.host:111: scripts/dtc/dtc] Error 1
> make[2]: *** [scripts/Makefile.build:585: scripts/dtc] Error 2
> make[2]: *** Waiting for unfinished jobs....
>   HOSTCC  scripts/mod/file2alias.o
>   SYSNR   arch/arm/include/generated/asm/unistd-nr.h
>   GEN     arch/arm/include/generated/asm/mach-types.h
>   SYSTBL  arch/arm/include/generated/calls-oabi.S
>   SYSTBL  arch/arm/include/generated/calls-eabi.S
>   CC      kernel/bounds.s
>   CHK     include/generated/timeconst.h
>   UPD     include/generated/timeconst.h
>   CHK     include/generated/bounds.h
>   UPD     include/generated/bounds.h
>   CC      arch/arm/kernel/asm-offsets.s
>   HOSTLD  scripts/mod/modpost
> make[1]: *** [Makefile:572: scripts] Error 2
> make[1]: *** Waiting for unfinished jobs....

RobertCNelson, 2020-06-04:
@blueForestIcarus sorry, this was probally fixed after 4.14.108, if you can find the stable SHA's from the v4.14.x mainline tree i'll glady add them.

Just note, TI has moved to v5.4 (after v4.19) and we will also be moving to v5.4 sooner..

Regards,

RobertCNelson, 2020-06-04:
Maybe? https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=v4.14.183&id=88f7a6aa7fb9aa5076b65489146045dac865f1d3

blueForestIcarus, 2020-06-04:
Is there a 5.4 xenomai kernel?

blueForestIcarus, 2020-06-04:
That patch you linked seems to be the right one, I'll try it and see.

blueForestIcarus, 2020-06-04:
Confirmed, that patch corrects it.