RobertCNelson/ti-linux-kernel-dev #41: can the branch 4.14.108-ti-xenomai-r136 be used on beaglebone ai?

lichaowuwei, 2020-07-22:
can this kernel be used on beaglebone ai? 
thanks!

and when i git clone the code, always be asked for the username of gitlab.denx.de, why?

RobertCNelson, 2020-07-22:
@lichaowuwei untested, give it a shot..

can you post the actual error you got in the terminal?

Regards,

lichaowuwei, 2020-07-23:
> @lichaowuwei untested, give it a shot..
> 
> can you post the actual error you got in the terminal?
> 
> Regards,

OK，i will  try it.  when i use the branch ti-Linux-xenomai-2.6.x   git://git.xenomai.org/xenomai-2.6.git  in git.sh can not find files，how to deal with  it

loserisme, 2022-05-16:
I did it on beaglebone ai.it can work,but icannot use wlan0

RobertCNelson, 2022-05-16:
brcmfmac has known issues with the RT kernel, wouldn't be surprised it was also broken on xenomai..

giuliomoro, 2022-06-18:
> I did it on beaglebone ai.it can work,but icannot use wlan0

@loserisme do you mean you got Xenomai to work on a BBAI? Is it the original BBAI (AM5729) or the new BBAI-64? What branch of this repo are you using? Did you make any changes to it to make it work? Last I tried, 4.14 was crashing on boot and required a soft restart to properly boot.

RobertCNelson, 2022-06-18:
This works be am57xx