RobertCNelson/ti-linux-kernel-dev #42: kernel 4.4 can't boot on beaglebone ai

lichaowuwei, 2020-07-30:
kernel 4.14-xenomai boot ok, 4.4 failed to boot.
can any modification to boot 4.4 kernel?
because of some reason, i want to use 4.4 kernel. why it can't boot?
Thanks very much and regards

lichaowuwei, 2020-08-03:
> Hi @lichaowuwei , this is correct, the old 4.4 kernel doesn't have the device tree changes required for the beaglebone ai..
> 
> at the time of the BBAI release, i was only supporting v4.14.x+ kernels
> 
> Sorry, but feel free to back port the device tree.
> 
> Regards,

how to  modify the  device  tree?any  instruction？

lichaowuwei, 2020-08-03:
> Hi @lichaowuwei , this is correct, the old 4.4 kernel doesn't have the device tree changes required for the beaglebone ai..
> 
> at the time of the BBAI release, i was only supporting v4.14.x+ kernels
> 
> Sorry, but feel free to back port the device tree.
> 
> Regards,

@RobertCNelson hi, why kernel 4.14 can't send and recieve ethercat frame? 4.4 kernel is ok! i printk some message, it seems like that when using 4.14kernel, no frame recieved, and cpsw_rx_handler never be executed! and dmesg say device eth0 entered promiscuous mode. when it not connect with ethercat device, dmesg indicates nomarlly. any thind wrong with the 4.14 kernel? can't use it with ethercat?

RobertCNelson, 2020-08-03:
hi @lichaowuwei sorry, i haven't used ethercat, are we missing a config, or ?

Regards,

lichaowuwei, 2020-08-03:
@RobertCNelson i don't know, it seems like many guys happen to this situation when use beaglebone with ethercat. or any where could try to config ? when downgrade to 4.4 or 3.8, things would be ok. but beaglebone ai could only use 4.14now, i can't boot it with 4.4 or don't know how to config the device tree.