RobertCNelson/ti-linux-kernel-dev #43: beaglebone AI errors in kernel and no wlan0

lowkin18, 2020-08-24:
Hello, I am running through the (BeagleBone AI) https://www.digikey.com/eewiki/display/linuxonarm/BeagleBone-AI tutorial.
When I follow the tutorial I get errors in the kernel and also wlan0 doesn't seem to work at all.

Just hoping you might happen to know the reason or how I can fix this. I am relatively new to linux and this is my first attempt at embedded linux. 

these are the errors I get in dmesg.
 omapdss error: HDMI I2C Master Error
dwc3 488d0000.usb: failed to initialize core

sdhci-omap 4809c000.mmc: Got CD GPIO
[    1.646216] sdhci-omap 4809c000.mmc: failed to set system capabilities
[    1.653377] sdhci-omap 480b4000.mmc: allocated mmc-pwrseq
[    1.658910] sdhci-omap 480b4000.mmc: failed to set system capabilities
[    1.665819] sdhci-omap 480d1000.mmc: allocated mmc-pwrseq
[    1.671348] sdhci-omap 480d1000.mmc: 480d1000.mmc supply vqmmc not found, using dummy regulator

[    7.883142] omap-rproc 58820000.ipu: assigned reserved memory node ipu1-memory@9d000000
[    7.927523] remoteproc remoteproc0: 58820000.ipu is available
[    7.944032] omap-rproc 55020000.ipu: assigned reserved memory node ipu2-memory@95800000
[    7.954090] remoteproc remoteproc0: Direct firmware load for dra7-ipu1-fw.xem4 failed with error -2
[    7.965249] remoteproc remoteproc1: 55020000.ipu is available
[    7.967424] remoteproc remoteproc0: powering up 58820000.ipu
[    7.973785] remoteproc remoteproc1: Direct firmware load for dra7-ipu2-fw.xem4 failed with error -2
[    7.978024] remoteproc remoteproc0: Direct firmware load for dra7-ipu1-fw.xem4 failed with error -2
[    7.990656] omap-rproc 40800000.dsp: assigned reserved memory node dsp1-memory@99000000
[    8.003994] remoteproc remoteproc1: powering up 55020000.ipu
[    8.004113] remoteproc remoteproc0: request_firmware failed: -2
[    8.009861] remoteproc remoteproc1: Direct firmware load for dra7-ipu2-fw.xem4 failed with error -2
[    8.027659] remoteproc remoteproc1: request_firmware failed: -2
[    8.045054] remoteproc remoteproc2: 40800000.dsp is available
[    8.070195] remoteproc remoteproc2: Direct firmware load for dra7-dsp1-fw.xe66 failed with error -2
[    8.079970] omap-rproc 41000000.dsp: assigned reserved memory node dsp2-memory@9f000000
[    8.090153] remoteproc remoteproc2: powering up 40800000.dsp
[    8.096739] remoteproc remoteproc2: Direct firmware load for dra7-dsp1-fw.xe66 failed with error -2
[    8.106702] remoteproc remoteproc2: request_firmware failed: -2
[    8.122029] remoteproc remoteproc3: 41000000.dsp is available
[    8.140667] remoteproc remoteproc3: Direct firmware load for dra7-dsp2-fw.xe66 failed with error -2
[    8.152653] remoteproc remoteproc3: powering up 41000000.dsp
[    8.158449] remoteproc remoteproc3: Direct firmware load for dra7-dsp2-fw.xe66 failed with error -2
[    8.175428] remoteproc remoteproc3: request_firmware failed: -2

[   10.712218] vpe 489d0000.vpe: Direct firmware load for vpdma-1b8.bin failed with error -2
[   10.725145] vpe 489d0000.vpe: couldn't get firmware

[   11.787482] cpu cpu0: _get_optimal_vdd_voltage: Failed optimized voltage match for 860000
[   11.811472] cpu cpu0: _get_optimal_vdd_voltage: Failed optimized voltage match for 860000

lowkin18, 2020-08-27:
Is this normal behavior?
I have run through the tutorial a few times now from different base machines and using different branches. 
I seem to have the same resulting kernel errors, which don't show up when using the supplied images from beaglebone.

 Also wondering if there is somewhere else I should be directing my questions.

Cheers,

Chris

RobertCNelson, 2020-08-27:
Are you booting the "ti-linux-4.14.y" or "ti-linux-4.19.y" branch?

All those "firmware" messages are normal, you are actually  free to write and load your own firmware blobs into those sub-devices.

In the BeagleBoard default images, we load special OpenCL blobs into the DSP to provide OpenCL functionality.

WiFi really only works great on v4.14 and v4.19 due to Cypress's brcmfmac patches..

make sure the "bb-bbai-firmware" package is installed for firwmare.

Regards,

lowkin18, 2020-08-28:
I have tried booting rt-4.14, rt-5.4, rt-4.9, 4.14, 4.9 , 5.4 and then xenomai 4.14. I never seem to have wlan0 and I don't know how to get it. 
I am going to try rt-4.19 tonight see if I get any other results.

I don't really know what you mean make sure the firmware package "bb-bbai-firmware" is installed is this post doing the tutorial? or during?  I have gone to the opt/scripts and do the copy.sh of the firmware for the bbai.

RobertCNelson, 2020-08-28:
@lowkin18 so none of the rt's work with brcmfmac patcheset, hardlock on bootup:

Grab teh v4.19.x-ti branch, it's the best tested to date..

```
sudo apt update
sudo apt install bb-bbai-firmware
```

Regards,

lowkin18, 2020-08-31:
I was able to get wlan0 working on 4.14 and 4.19 and 5.4, just needed to rfkill unblock all.

Just wondering if getting wlan0 working with the rt branches is in the pipeline or if it's possible to fix myself.

Thanks you for the help

lowkin18, 2020-08-31:
Hello,

I was able to boot 4.14, 4.19 and 5.4 and get WLAN0 working on all of them. Just user error on my part, needed to set WLAN0 up. Thank you for your help.

EtherCat with SOEM doesn't seem to work. I see someone else was saying that the kernel introduced something that makes it fail to read. Just wondering if something that fixes this is in the pipeline or if there is a branch that is known to work.

Thanks Again.

lowkin18, 2020-08-31:
Well I've managed to muck the flow of this, sorry.

I was able to get WLAN0 on 4.14,4.19 and 5.4 working, just need to unblock it with RFKILL

However the wlan0 doesn't seem to work on boot if the eth0 is plugged in. Wondering if this is a known issue or just something I mucked up.

lowkin18, 2020-09-02:
Thank you very much for the help and information.

closed 2020-09-02 by lowkin18