RobertCNelson/ti-linux-kernel-dev #44: How to update to latest 4.19.y ?

JoseGoncalves, 2020-09-01:
I'm dealing with an issue in my application and would like to test it with the latest 4.19.y kernel (currently, 4.19.142).
Can you tell me how can I build a kernel based on the latest mainline 4.19.y and with the BeagleBone 's specific patches?
The naife approach of replacing in `version.sh` `KERNEL_TAG=${KERNEL_REL}.94` with `KERNEL_TAG=${KERNEL_REL}.142` does not work, as it generates a lot of patches conflicts.

RobertCNelson, 2020-09-01:
The "ti" branch ends at 4.19.94:

https://git.ti.com/gitweb/?p=ti-linux-kernel/ti-linux-kernel.git;a=shortlog;h=refs/heads/ti-linux-4.19.y

Switch to the bone v4.19.x branch

```
sudo apt install linux-image-4.19.135-bone55
```

Regards,

JoseGoncalves, 2020-09-01:
In that case, how can I build an image based in 4.9.142 on the bone 4.19.x branch?

RobertCNelson, 2020-09-01:
Feel free to change it in version.sh

https://github.com/RobertCNelson/bb-kernel/tree/am33x-v4.19

It's only 4.19.135 to keep the RT and non-RT in sync:

https://mirrors.edge.kernel.org/pub/linux/kernel/projects/rt/4.19/

Regards,

closed 2020-09-01 by JoseGoncalves