RobertCNelson/ti-linux-kernel-dev #45: Beaglebone AI and I suspect all the BB don't work with etherCat

lowkin18, 2020-09-03:
Hello,
I was unable to get Simple Open EtherCAT master on the BBAI on 4.14, 4.19 and 5.4. I brought this up with the guys in charge of the SOEM repository and they linked me to this potential issue.

https://patchwork.kernel.org/patch/10697191/

The CPSW ethernet hardware is padding the packets wrong, which is causing issue with some industrial automation protocols.
Such as my experience with a EtherCAT slave not responding to packets sent form the BBAI.

I hope this is the correct location to post this.

Cheers,

Chris

closed 2020-09-03 by lowkin18

RobertCNelson, 2020-09-03:
@lowkin18 please keep us updated, we'd like to see ethercat working, but dont' have the hardware to verify it..

lowkin18, 2020-09-04:
> @lowkin18 please keep us updated, we'd like to see ethercat working, but dont' have the hardware to verify it..

Hello Robert,

I was able to get it working by making a change to tx_packet_min=60 in cpsw.c locate at "ti-linux-kernel-dev/KERNEL/drivers/net/ethernet/ti/cpsw.c"

I am now trying to do it using the built module parameter that is in the 5.4 version
```
static int tx_packet_min = CPSW_MIN_PACKET_SIZE;
module_param(tx_packet_min,int,0444);
MODULE_PARM_DESC(tx_packet_min, "minimum tx packet size (bytes)");
```
I however am relatively new to embedded linux and don't really know the correct way to change a module parameter that is baked into the kernel. I will report back once I get it working through this module parameter way.

lowkin18, 2020-09-04:
I am struggling to find where I would setup the kernel cmdline to pass in the ti_cpsw.tx_packet_min=60
Is this a file that lives in ti-linux-kernel-dev or something I am supposed to pipe in from u-boot.