RobertCNelson/ti-linux-kernel-dev #50: no MDIO, no PHY, no Ethernet on BeagleBoneBlack with Kernel >5 (tested on 5.4 and 5.10)

Rockstein2, 2021-08-14:
Sorry if already there. Google could not really help me. Everything was out of date.
On my BeagleBoneBlack Board Rev. B6 with AM3358BZCZ100 the MDIO does not come up with kernel 5.4 and 5.10. 

dmesg says  following
**mdio_bus 4a101000.mdio: MDIO device at address 0 is missing
phy "/ocp/interconnect@4a0000/segment@0/target-module@100000/ethernet@0/mdio@1000/ethernet-phy@0" not found on slave 0**

I changed in /arch/arm/boot/dts/am335x-bone-common.dtsi under "ethphy0: ethernet-phy@0" _reg = <0>_ to **_reg = <2>_**. After that everything was ok. I am not sure if I have fixed it correctly. Maybe it is better to overlay it in am335x-boneblack.dts because the am335x-bone-common.dtsi is included by several boards. 

It took me a while to find it.  

Again the correct section am335x-bone-common.dtsi 
&davinci_mdio {
	pinctrl-names = "default", "sleep";
	pinctrl-0 = <&davinci_mdio_default>;
	pinctrl-1 = <&davinci_mdio_sleep>;
	status = "okay";

	ethphy0: ethernet-phy@0 {
		reg = <2>;
	};

RobertCNelson, 2021-08-14:
Actually, it is zero..... The phy is buggy, there is a patch in the tree which tries to rediscover it, but it's broken with ti latest changes.. what uname -r are to booting with.. there is a better fix in uboot I'm going to start moving to that.

Rockstein2, 2021-08-14:
Thanks for the quick reply. Surprisingly it still worked with 5.4.40 from a year ago. 
By "PHY is buggy" you mean the power on problem, which comes very rarely after PowerUp?   It is due to duration of SYS_RESET pulse and I have seen somewhere a HW solution with capacitor. 
Anyway thanks for your great work.

boneblackbeagle, 2024-04-10:
I have this issue with 5.10.168-ti-r77.
On U-boot I see : Net:   eth2: ethernet@4a100000, eth3: usb_ether
I can fix the issue with the following in the debug terminal : sudo gpioset gpiochip1 8=0
After this command the led-s of the Ethernet starts blinking and after a reboot I have working network.

Maybe the reset-deassert-us = <6500>; is not enough in the am335x-bone-common.dtsi ?

Another interesting thing is that with 5.10.168-ti-r72 works well all the time.

boneblackbeagle, 2024-04-11:
I have changed the reset-deassert-us = <6500>; to reset-deassert-us = <10500>; and still working after about a 20 cycle with 1 min off time between two startup.