RobertCNelson/ti-linux-kernel-dev #54: Using rpmsg crashes the drive

snppla, 2023-09-22:
As an example of using the rpmsg feature, this application (https://github.com/RoSchmi/Beaglebone-PRU-RPMsg-HelloWorld ) demonstrates what I'm seeing when I'm trying to use rpmsg.

Once I build main0.c for the pru, I copy am335x-pru0-fw  into /lib/firmware, and start it with `echo start >> /sys/class/remoteproc/remoteproc1/state` I get the following error in dmesg (then the system becomes unstable until I reboot).

```
[ 2215.373274] remoteproc remoteproc1: powering up 4a334000.pru
[ 2215.390174] remoteproc remoteproc1: Booting fw image am335x-pru0-fw, size 74304
[ 2215.390416] remoteproc remoteproc1: unsupported resource 5
[ 2215.394040] pru-rproc 4a334000.pru: Allocated carveout doesn't fit device address request
[ 2215.394208] pru-rproc 4a334000.pru: Allocated carveout doesn't fit device address request
[ 2215.399406]  remoteproc1#vdev0buffer: registered virtio0 (type 7)
[ 2215.399438] remoteproc remoteproc1: remote processor 4a334000.pru is now up
[ 2215.474222] 8<--- cut here ---
[ 2215.483349] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[ 2215.502608] pgd = f4da8766
[ 2215.506508] [00000000] *pgd=00000000
[ 2215.511369] Internal error: Oops: 5 [#1] PREEMPT SMP ARM
[ 2215.516922] Modules linked in: virtio_rpmsg_bus(+) rpmsg_core ip6t_REJECT nf_reject_ipv6 ipt_REJECT nf_reject_ipv4 ip6table_filter ip6_tables xt_MASQUERADE nf_conntrack_netlink nfnetlink xfrm_user xfrm_algo br_netfilter wl18xx wlcore mac80211 cfg80211 xt_owner libarc4 hci_uart btqca btrtl btbcm btintel rfcomm bluetooth pru_rproc irq_pruss_intc pm33xx pruss ti_eqep counter c_can_platform c_can can_dev spidev evdev wkup_m3_ipc uio_pdrv_genirq uio cpufreq_dt sch_fq_codel
[ 2215.560312] CPU: 0 PID: 3094 Comm: systemd-udevd Not tainted 5.10.162+ #1
[ 2215.567384] Hardware name: Generic AM33XX (Flattened Device Tree)
[ 2215.573750] PC is at pru_rproc_kick+0x54/0x100 [pru_rproc]
[ 2215.579478] LR is at rproc_virtio_notify+0x2c/0x54
[ 2215.584469] pc : [<bf062584>]    lr : [<c0c305b0>]    psr: 20030013
[ 2215.590999] sp : d009dcf0  ip : d009dcf8  fp : dc909e00
[ 2215.596436] r10: 00004000  r9 : dc908000  r8 : 00000cc0
[ 2215.601877] r7 : 00000000  r6 : d00b3c00  r5 : c4017420  r4 : c40176f0
[ 2215.608675] r3 : 00000000  r2 : 00000001  r1 : 00000000  r0 : bf064770
[ 2215.615474] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[ 2215.622901] Control: 10c5387d  Table: 9000c019  DAC: 00000051
[ 2215.628888] Process systemd-udevd (pid: 3094, stack limit = 0x2cfb087a)
[ 2215.635778] Stack: (0xd009dcf0 to 0xd009e000)
[ 2215.640320] dce0:                                     00000cc0 bf33418c bf064770 bf064774
[ 2215.648841] dd00: bf064778 c9168545 00000cc0 c4017400 00000000 c0c305b0 d0235d80 00000001
[ 2215.657363] dd20: d00b3c00 c08804a0 d0235180 bf334a28 00000035 00000000 00000000 c03f7ae4
[ 2215.665881] dd40: d0282ec8 bf3360f8 00000000 dfd65546 00000e00 00000200 00000000 bf334c94
[ 2215.674400] dd60: bf334028 d0235d80 d0235300 c9168545 d0282cb8 d00b3c10 d00b3c00 d00b3e00
[ 2215.682921] dd80: bf338000 00000000 00000000 00000001 00000000 c0880368 d00b3c10 00000000
[ 2215.691441] dda0: 00000000 d00b3c10 c298a82c 00000000 00000000 c298a834 bf338000 00000023
[ 2215.699958] ddc0: 02352cb8 c09753d0 d00b3c10 c9168545 02352cb8 d00b3c10 bf338000 d00b3c54
[ 2215.708480] dde0: bf338000 d0235938 d009c000 0000017b 02352cb8 c0975bf8 00000000 c9168545
[ 2215.716998] de00: d009c000 d00b3c10 00000000 d00b3c54 bf338000 d0235938 d009c000 c0975fa8
[ 2215.725519] de20: 00000000 bf338000 d00b3c10 c28b1fe4 d0235938 c097605c bf338000 bf338000
[ 2215.734038] de40: c0975fb0 c28b1fe4 d0235938 c0972fc4 c309716c c3097158 d02354b4 c9168545
[ 2215.742558] de60: bf338000 d0235900 00000000 c0974600 bf336530 c08072a8 bf338000 00000000
[ 2215.751077] de80: 0000000f 00000000 c01002c4 c0976be8 bf33b000 d0130bc0 0000000f bf33b010
[ 2215.759600] dea0: bf33b000 c0102170 00000cc0 c280cd78 c01f43f4 00000008 00000000 b03f185d
[ 2215.768118] dec0: d009dec8 c0336fc8 d009dee0 00000000 80270d0d b03f185d 00000cc0 c01f43f4
[ 2215.776642] dee0: d0130bc0 c9168545 00000000 bf3381c0 b6ccb620 c9168545 0000017b bf3381c0
[ 2215.785162] df00: d0130bc0 0000000f 0000017b c01f4414 0000017b c01002c4 00000000 b6ccb620
[ 2215.793685] df20: 0000000f c01f7558 d009df3c 7fffffff 00000000 00000002 00000000 e24d6000
[ 2215.802205] df40: e24d8081 e24d77c0 e24d6000 00005b68 e24db4d8 e24db338 e24d9de4 00005000
[ 2215.810726] df60: 000052e0 000036fc 00005523 bf338090 00000008 00000000 000036ec 00000027
[ 2215.819245] df80: 00000028 0000000f 00000021 00000020 00000000 c9168545 00000000 00000000
[ 2215.827762] dfa0: 00000000 c01002a0 00000000 00000000 0000000f b6ccb620 00000000 00020000
[ 2215.836283] dfc0: 00000000 00000000 00000000 0000017b 00000000 b6e8137b 00000000 02352cb8
[ 2215.844802] dfe0: be90a790 be90a780 b6cc65b3 b6bfaf22 40010030 0000000f 00000000 00000000
[ 2215.853346] [<bf062584>] (pru_rproc_kick [pru_rproc]) from [<c0c305b0>] (rproc_virtio_notify+0x2c/0x54)
[ 2215.863150] [<c0c305b0>] (rproc_virtio_notify) from [<c08804a0>] (virtqueue_notify+0x1c/0x34)
[ 2215.872064] [<c08804a0>] (virtqueue_notify) from [<bf334a28>] (rpmsg_probe+0x2a8/0x380 [virtio_rpmsg_bus])
[ 2215.882147] [<bf334a28>] (rpmsg_probe [virtio_rpmsg_bus]) from [<c0880368>] (virtio_dev_probe+0x220/0x2e0)
[ 2215.892211] [<c0880368>] (virtio_dev_probe) from [<c09753d0>] (really_probe+0xf8/0x4e4)
[ 2215.900553] [<c09753d0>] (really_probe) from [<c0975bf8>] (driver_probe_device+0xf4/0x1c8)
[ 2215.909163] [<c0975bf8>] (driver_probe_device) from [<c0975fa8>] (device_driver_attach+0xa8/0xb0)
[ 2215.918409] [<c0975fa8>] (device_driver_attach) from [<c097605c>] (__driver_attach+0xac/0x144)
[ 2215.927379] [<c097605c>] (__driver_attach) from [<c0972fc4>] (bus_for_each_dev+0x74/0xc0)
[ 2215.935899] [<c0972fc4>] (bus_for_each_dev) from [<c0974600>] (bus_add_driver+0x114/0x208)
[ 2215.944511] [<c0974600>] (bus_add_driver) from [<c0976be8>] (driver_register+0x88/0x118)
[ 2215.952948] [<c0976be8>] (driver_register) from [<bf33b010>] (rpmsg_init+0x10/0x1000 [virtio_rpmsg_bus])
[ 2215.962843] [<bf33b010>] (rpmsg_init [virtio_rpmsg_bus]) from [<c0102170>] (do_one_initcall+0x40/0x2cc)
[ 2215.972642] [<c0102170>] (do_one_initcall) from [<c01f4414>] (do_init_module+0x44/0x254)
[ 2215.981074] [<c01f4414>] (do_init_module) from [<c01f7558>] (sys_finit_module+0xbc/0x120)
[ 2215.989596] [<c01f7558>] (sys_finit_module) from [<c01002a0>] (__sys_trace_return+0x0/0x20)
[ 2215.998293] Exception stack(0xd009dfa8 to 0xd009dff0)
[ 2216.003555] dfa0:                   00000000 00000000 0000000f b6ccb620 00000000 00020000
[ 2216.012075] dfc0: 00000000 00000000 00000000 0000017b 00000000 b6e8137b 00000000 02352cb8
[ 2216.020591] dfe0: be90a790 be90a780 b6cc65b3 b6bfaf22
[ 2216.025856] Code: e320f000 e5943054 e3a02001 e3a01000 (e5930000)
[ 2216.039295] ---[ end trace fd9f2024eca60239 ]---

```

Output from uname 
`Linux 22018b9 5.10.162+ #1 SMP PREEMPT Fri Apr 7 00:25:14 UTC 2023 armv7l armv7l armv7l GNU/Linux`

Kernel tag: 5.10.162-ti-r59

uEnv.txt (the overlays are put into a different location).

```
enable_uboot_overlays=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
disable_uboot_overlay_wireless=1
disable_uboot_overlay_adc=1
uboot_overlay_pru=/boot/overlays/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo
cmdline=coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet
```

RobertCNelson, 2023-09-22:
@snppla sadly, TI rewrote the remoteproc stack with each major release, the demo you reference was tested with.... `4.4.91-ti-r133`... yeah, it's not going to work on 5.10..

Regards,

snppla, 2023-09-22:
That explains why it won't work. If you don't have an immediate direction to go, I'll go about scouring the docs & other examples to try something else. Any idea if the device tree overlay would also need to be changed with the changing stack?

RobertCNelson, 2023-09-22:
@snppla 6.0.0 and greater of this ti project https://git.ti.com/gitweb?p=pru-software-support-package/pru-software-support-package.git;a=summary supports 5.10.x remoteproc...

snppla, 2023-09-22:
I just tried it out, works as expected. Thanks!

closed 2023-09-22 by snppla